adafruit/circuitpython #3841: 6.0.0 not working for me on Fomu

tcal-x, 2020-12-17:
I downloaded `adafruit-circuitpython-fomu-en_US-6.0.0.dfu` from `https://circuitpython.org/board/fomu/` and installed it on Fomu with dfu-util 0.9.

The RGB LED gives a nice pinkish breathe/blink, but no tty device shows up in /dev, neither ttyUSB nor ttyACM.   

I've tried with foboot v2.0.3 and v2.0.3-55-ge928b8,  two different Fomus, with the same results.

`dmesg` just shows `USB disconnect, device number xxx`.

`adafruit-circuitpython-fomu-en_US-5.3.1.dfu` works correctly on the same Fomus.

tcal-x, 2020-12-17:
I built circuitpython at `tag: 5.4.0-beta.1` (`0ba541d49878f8fa62aca7822e8c7d333dd5c2f0`, Jun 8 2020) and that did work on Fomu.

tcal-x, 2020-12-17:
Commit `f232aef7865c4e094f060f9766418f60fb9f86ea` (Jun 15, 2020) "supervisor.mk: Compute USB_DEVICES; remove from boards and ports" also WORKS.

tcal-x, 2020-12-17:
f61c4e62c14e52da83b17f740b5c5409ee4b05a5 (Nov 11, 2020) -- not working
3dfed3c4aa52d06bd2eb1c28d0a1916821cc3f80 (Jul 24, 2020) -- not working
2d579cc9953a3ba965973c6d5139c007b3c1b38d (Jun 26, 2020) -- works

dhalbert, 2020-12-18:
@xobs would you like to take a look at this?

xobs, 2020-12-22:
Sure, let me take a look at it this evening.

xobs, 2020-12-24:
I have it hooked up to an EVT board, and it's looking like it's not properly clearing interrupts. In particular, whenever I break into the debugger it's calling `CALLBACK_CRITICAL_END;` in `background_callback.c:57`.

xobs, 2020-12-24:
It's leaking stack somehow. Every time an interrupt hits, the stack pointer is moved up by 160 bytes.

xobs, 2020-12-24:
There were two issues at play:

1. Interrupts were getting re-enabled prior to the handler returning. This happened both in `port_get_raw_ticks()` and in `supervisor_tick()`.
2. Some code was placed in the critical path, resulting in timeouts from the USB stack. Notably, a `divide` or `modulo` operation was added somewhere along the line

xobs, 2020-12-24:
Github appears to be broken, and can't create a pull request:

[image]

Please apply https://github.com/xobs/micropython/commit/f3e54414e5771a2463a8a93b4dcbb395584d2fe3 and https://github.com/xobs/micropython/commit/2f95cc95a494395658eefed6130b9a8732117482 to fix this issue

fancellu, 2021-01-27:
Ditto for me, neither 6.1.0 or 6.2.0-beta.1, they get to the end, no errors show, but no rainbow and fomu flashes red, no usb drive appears

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2019 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1209
Match product ID from file: 5bf0
Opening DFU capable USB device...
ID 1209:5bf0
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 4096
Copying data from PC to DFU device
Download        [=========================] 100%       522288 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
```

dfu-util -l says

```
Found DFU: [1209:5bf0] ver=0101, devnum=3, cfg=1, intf=0, path="2-1.4", alt=0, name="Fomu PVT running DFU Bootloader v2.0.3", serial="UNKNOWN"
```

tannewt, 2021-01-28:
@xobs can you make a PR now? If you get the unicorn again please share the full url. I get this if I'm on the wrong base branch.

dhalbert, 2021-12-24:
I think this was fixed by #3878 (a long time ago).

closed 2021-12-24 by dhalbert