adafruit/circuitpython #3843: JTAG stopped working on Kaluga

jepler, 2020-12-17:
I recently experienced that shortly after starting CircuitPython, JTAG debugging on the Kaluga would stop.  I think this may be due to #3643, though I didn't bisect it.  I can work around it by making the jtag pins never-reset, but I'm not sure if this is the right fix since this (I think) would stop people using them as GPIO in general.
```
+    common_hal_never_reset_pin(&pin_GPIO39);
+    common_hal_never_reset_pin(&pin_GPIO40);
+    common_hal_never_reset_pin(&pin_GPIO41);
+    common_hal_never_reset_pin(&pin_GPIO42);
```
Also, the existing never_reset calls are in a board-specific file so this would need to be carried across all boards.

Is there a better way?  Or should I go ahead and submit a PR along these lines?

tannewt, 2020-12-22:
I was hoping the new reset code would preserve it by leaving the mux alone. I think the right fix is to preserve any pulls as well on those pins.

dhalbert, 2026-11-13:
Claude wrote this:

This should be fixed. In an `ENABLE_JTAG=1` build, `port_init()` now claims the JTAG pins at startup (#11755, in 11.0.0-alpha.3). User code gets "pin in use" if it tries to take them. The same PR removed the reset of all pins between runs of user code, so pins are reset only when the object using them is deinited. No object owns the JTAG pins, so nothing resets them, and their pin function and pulls stay as they were. (Before that, starting with #3733 in 6.1.0, `DEBUG` and `ENABLE_JTAG` builds marked them never-reset.)

Claude tested `main` (`ab606c9484`) with `ENABLE_JTAG=1` on a Metro ESP32-S3 (GPIO39–42) and a Metro ESP32-P4 (GPIO2–5). All four pins were in use after boot, after a reload, and after a USB power cycle. Without `ENABLE_JTAG`, they were free.

Claude did not test a live JTAG session on these pins. By default, the S3 and P4 route JTAG to their built-in USB-JTAG bridge instead.

closed 2026-11-13 by dhalbert