adafruit/circuitpython #3844: Auto-reload not working during simulated deep sleep?

dhalbert, 2020-12-18:
From a feedback comment on the new Deep Sleep Guide:

>One thing I've noticed about using deep sleep that was very confusing at first (as well as regular sleep for long periods) is that code no longer auto-reloads when I save in Mu. It seems like the Magtag is hanging. You need to mouse into the REPL and type CTRL-C.

ladyada, 2020-12-18:
yes, this bug report is reproducable - i have to manually restart after updating code

tannewt, 2020-12-22:
This might be fixed in my PinAlarm PR. It tweaks how we notify the CP task from other tasks.

dhalbert, 2021-03-17:
I retested this with a simple program on the MagTag, and I don't see this problem now.
```py
import alarm
import time

print("HelloWorld!")   # I edited this during simulated deep sleep.

a = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 10)
alarm.exit_and_deep_sleep_until_alarms(a)
```

closed 2021-03-17 by dhalbert