adafruit/circuitpython #3845: bitbangio.I2C does not work

jepler, 2020-12-18:
On the Kaluga, I have two sensors attached to the I2C bus at IO10/11.  They are detected if I scan with busio.I2C but not with bitbangio.I2C.  When I scope bitbangio, there is literally no activity on the pins 10/11; there is the expected activity with busio.I2C.

```
>>> import board,bitbangio
>>> bus = bitbangio.I2C(board.IO10, board.IO11); bus.try_lock(); bus.scan(); bus.deinit()
True
[]
>>> import busio, board
>>> bus = busio.I2C(board.IO10, board.IO11); bus.try_lock(); bus.scan(); bus.deinit()
True
[11, 112]
```

I'm testing a local build.  Possibly important, this is a `DEBUG=1` build.

tannewt, 2020-12-22:
Maybe we aren't correctly supporting open drain outputs in the `digitalio` code.

jepler, 2020-12-23:
I tested with `adafruit_bitbangio` and that works on Kaluga. However, on further inspection, `adafruit_bitbangio.I2C` doesn't seem to use `drive_mode=digitalio.DriveMode.OPEN_DRAIN` which is a surprise!  I thought that would answer the open drain question but I guess it doesn't.

jepler, 2020-12-23:
I wired up a couple of LEDs and can confirm that open drain doesn't work, or at least not as I expect it.

In push/pull mode, one LED lights brightly depending whether the pin's value is True or False.

In open drain mode, neither LED lights no matter whether the value is True or False.  However, the upper LED should light when value is False.

[image]

I did verify that in input mode, pull up/down/none all work; the pulls can be seen as a dimly lighting LED.

dhalbert, 2022-01-16:
Simple fix for this, which I am doing as part of #5816:
```patch
@@ -108,11 +108,9 @@ digitalinout_result_t common_hal_digitalio_digitalinout_set_drive_mode(
     digitalio_digitalinout_obj_t *self,
     digitalio_drive_mode_t drive_mode) {
     gpio_num_t number = self->pin->number;
-    gpio_mode_t mode;
+    gpio_mode_t mode = GPIO_MODE_DEF_OUTPUT;
     if (drive_mode == DRIVE_MODE_OPEN_DRAIN) {
-        mode = GPIO_MODE_DEF_OD;
-    } else {
-        mode = GPIO_MODE_DEF_OUTPUT;
+        mode |= GPIO_MODE_DEF_OD;
     }
     esp_err_t result = gpio_set_direction(number, mode);
     if (result != ESP_OK) {
```

PR #5866 (2022-01-17): fix espressif DigitalInOut open-drain
- Fixes #3845.

Open-drain and output both need to be set. Discovered while researching #5865, which is not yet fixed.

Tested with a Saleae and a QT Py ESP32-S2.
closed 2022-01-17 by jepler