adafruit/circuitpython #3849: ESP32-S2 2nd USB-SERIAL or more interfaces than endpoinds with static descriptor

emard, 2020-12-19:
My basic wish is to have second USB-SERIAL device in
ESP32-S2 which will lead to hardware TX/RX lines.

I was thinking - if descriptor must be static, and list all supported
device for which same endpoints have to be shared, then
perhaps on the OS (at least linux) is possible to blacklist unneeded
drivers, so that actually endpoints will be used only by not blacklisted
drivers, and things may work. (at least in heory)

dhalbert, 2020-12-19:
I'm not quite sure what you're looking for. We have been thinking about a secondary CDC serial connection over USB (e.g. you'd have `/dev/tty/ACM0` for the REPL connection, and then also a `/dev/ttyACM1`. But if you want hardware UART, then you want to use `busio.UART`.

emard, 2020-12-19:
Yes I'd like to have this secondary CDC /dev/ttyACM1
and a possibility to bridge its traffic to external pins TX/RX

it would be practical to run usb-serial bridge in
"background" of esp32 so that more user code
can work too if technologically possible.

ladyada, 2020-12-19:
this may be a dup of 
https://github.com/adafruit/circuitpython/issues/231
?

dhalbert, 2020-12-19:
I rewrote this as a more general issue in #3853 and will close this in favor of that issue.

closed 2020-12-19 by dhalbert