adafruit/circuitpython #3851: MatrixPortal eventually hangs if serial data is not flushed

jldeon, 2020-12-19:
I've got a MatrixPortal board running CircuitPython which is having some issues.  If I connect the board to a PC (or any device that can handle the MatrixPortal's USB serial output), then the board runs fine for extended periods of time.  If I hook it up to a charger (or a PC that is sleeping, etc), then the board will eventually hang.

I've run a few experiments and been able to demonstrate that I can get the board out of the hung state by waking the PC it is connected to.  

Here's a simple sample to reproduce:

```
import time
count = 0
while True:
    print(count)
    for i in range(2000):
        print(count, end=" ")
    print("\nnaptime")
    time.sleep(300)
```

1. Plug MatrixPortal into a PC 
2. Upload the code snippet above
3. Monitor the serial output and confirm that the code is working properly
4. Put the PC to sleep 
5. Wait some length of time (hours produces the most obvious result)
6. Wake the PC and check the serial output

The expected result is that the "count" value printed should be approximately "the amount of time the board has been active, divided by five minutes."  In my experiments, the "count" value printed was much lower.  

The amount of loops that the board is capable of doing before it hangs can be changed by changing the amount of printing that is done on each loop.  (ie, change the constant in the `range(2000)` call).  This would seem to indicate that the issue is that the `print` statement is hanging, possibly due to a buffer being full?  I'm not that familiar with the lower level CircuitPython (or MatrixPortal support).

Cyphrz, 2021-01-22:
I'm able to confirm this issue as well.

dhalbert, 2021-03-31:
This _may_ be fixed by #4517, which was just merged. It is in "Absolute Newest" and will be in rc.0. Could you re-test?

jldeon, 2021-04-03:
So far in testing, this does seem to have improved in rc.0!  I will continue to test this weekend before I have to put my MatrixPortal back into "production mode" on Monday.  I'll reopen if I run into this issue again.

closed 2021-04-03 by jldeon