adafruit/circuitpython #3852: UMFEATHERS2 Dx pins are not consistent with other feathers

jerryneedell, 2020-12-19:
Several of the pin mappings in the pins.c file for the umfeathers2 are not constant with the other feather boards.

Notable are D5,D6,D9,D10  --  many in the "short side".
https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/boards/unexpectedmaker_feathers2/pins.c#L50

Since it is "feather" it would be nice to keep the names consistent.

Is there some reason why this is not desirable or possible for this board?
I'll be happy to create a PR to rename them if no one objects.

ladyada, 2020-12-19:
thats sorta up to @unexpectedmaker - its whatever they want for their board :)

anecdata, 2020-12-19:
I can't find an issue or PR, but there was an extensive discussion on Discord about the challenge of ESP32-S2 pin naming, in the context of Metro S2 (whose long-side pins don't match Metro M4, for example)

jerryneedell, 2020-12-19:
Sure. It's not a big deal, but I just put my umfeathers2 into  keyboard-feathering and have gotten most of the functions working OK but I had to do a lot of pin remapping that I think could be avoided. Also there is one hard code pin in the featherwing_keyboard library (D6) that won't work as is. Perhaps that should be fixed in the library and passed in as an argument like all the other pins.

I agree that for other boards, it may not be worthwhile, but since the "feather" layout is so common and since there are Dx names assigned, it would be nice to make them consistent.

anecdata, 2020-12-19:
Since `board` pins can have multiple names, there's a reasonable argument to be made to add the standard Feather naming to the names that match the silk, etc.

jerryneedell, 2020-12-19:
I put a PR in for the library https://github.com/adafruit/Adafruit_CircuitPython_FeatherWing/pull/66 and it works OK.

ahh - I see that the D-names do match those on the underside silkscreen so I guess that is how they should stay.
I had not looked at the bottom silkscreen and through they could be freely changed.
I'll close this.

closed 2020-12-19 by jerryneedell

tannewt, 2020-12-22:
Note that this alternative numbering is an artifact of ESP Arduino not being able to remap numbers. They don't have a layer of indirection for pin numbering (yet: https://github.com/espressif/arduino-esp32/issues/4479). This impacts CircuitPython when folks use the arduino numbering on the board itself.