adafruit/circuitpython #3853: Provide an optional second USB CDC serial channel

dhalbert, 2020-12-19:
Several other issues (#231, #3849, for example) want to communicate serially with a host computer without the REPL interfering. We have discussed providing a second USB CDC serial device. This probably needs dynamic USB descriptors (#1015), because of the shortage of endpoints, and the confusion that an always-present second CDC might present.

emard, 2020-12-19:
I suggest to test-implement 2nd CDC channel with static descriptor,
observe problems what happens with host os and fix problems then.

tannewt, 2020-12-22:
TinyUSB has a dual CDC example already here: https://github.com/hathach/tinyusb/tree/master/examples/device/cdc_dual_ports

ladyada, 2020-12-22:
during discussion today i asked if we could implement secondary CDC UART on one or two platforms that have the spare endpoints now, rather than blocking on a dynamic descriptor implementation

tannewt, 2020-12-22:
> during discussion today i asked if we could implement secondary CDC UART on one or two platforms that have the spare endpoints now, rather than blocking on a dynamic descriptor implementation

We certainly could. I think that'd get us most of the way to supporting dynamic descriptors. We have most of the prerequisites in place already.

kevinjwalters, 2021-01-26:
For reference, this may well be of interest for this user and anyone who finds this question: [How to do non blocking usb serial input in circuit python?](https://stackoverflow.com/questions/65647986/how-to-do-non-blocking-usb-serial-input-in-circuit-python/65736367).

ladyada, 2021-02-20:
added

closed 2021-02-20 by ladyada