adafruit/circuitpython #3859: adafruit_bus_device: SPI object doesn't have spi property (and adafruit_sdcard assumes it)

jepler, 2020-12-22:
Reported by @neradoc on Discord, this issue is separate from other problems encountered with `adafruit_bus_device`.

It wasn't immediately clear why adafruit_sdcard uses this property (efficiency?) but as the property was not `_spi` it was arguably fair game.  Possible resolutions that occurred to me:
 * change `adafruit_sdcard` to not use the property and leave in-core `adafruit_bus_device` unchanged for the purposes of this PR
 * add the property to `adafruit_bus_device` and leave `adafruit_sdcard` unchanged
 * change both even though it's redundant
I didn't find any uses of `adafruit_bus_device.i2c_device.I2C.i2c` with my quick grepping in the Learn system code and Bundle, but the same concern may apply there.

tannewt, 2020-12-29:
This no longer blocks 6.1.0 because we've disabled native bus device for now.

dhalbert, 2021-02-05:
Fixed by #4114.

closed 2021-02-05 by dhalbert