adafruit/circuitpython #3876: UART read can block indefinitely on nRF52840 6.0.0

kevinjwalters, 2020-12-24:
I've seen what looks like read() not returning on a UART creating with timeout 4 seconds on standard TX/RX on a Feather nRF52840 Express. The code gets stuck here:

```
Traceback (most recent call last):
  File "code.py", line 267, in <module>
  File "code.py", line 265, in <module>
  File "lib/pimoroni_pms5003/__init__.py", line 156, in read
KeyboardInterrupt:

Code done running. Waiting for reload.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0 on 2020-11-16; Adafruit Feather nRF52840 Express with nRF52840
```

It's reported as line 156 but line 155 (assuming I have matching code here) is where the call reall is

https://github.com/pimoroni/pms5003-circuitpython/blob/e3c4fa05fec7364fa23506afe17e0bc6f255a9d5/library/pms5003/__init__.py#L155-L156

_If all is well_ that's a read for 28 bytes, it's possible the frame is damaged so could be a read for any value between 0 and 65535.

Data is being sent here in 32 byte lumps at 9600 baud 8n1 roughly every 900ms.

That feather conveniently has the SWD connector. Would that be useful here?

dhalbert, 2020-12-24:
Could you also test on 5.3.1 if possible? That would help narrow it down. Thanks!

kevinjwalters, 2020-12-24:
I've done a little bit of REPL testing and it's possible this is actually just bogus frame lengths, e.g. something like a read(largenumber) could be creating the appearance of getting stuck.

I'm going to make the code more robust for abnormal frame lengths and then I'll study the behaviour afterwards to see if this still occurs. If it does I'll give `5.3.1` a go too, thanks.

kevinjwalters, 2021-02-01:
I'm thinking this was most likely to be `read(65535)` against a source producing about 36 bytes a second. I'll close this for now.

closed 2021-02-01 by kevinjwalters