adafruit/circuitpython #3884: microcontroller.reset() to bootloader on esp32-s2

FoamyGuy, 2020-12-27:
Running these commands in REPL my device is rebooting into standard mode coming up as a `CIRCUITPY` drive.
```
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
microcontroller.reset()
```
Tested on:
```
Adafruit CircuitPython 6.1.0-beta.2-182-g1b7fbaf31 on 2020-12-26; Adafruit MagTag with ESP32S2
```

microDev1, 2020-12-27:
`microcontroller.RunMode.BOOTLOADER` isn't implemented on the esp32s2.

https://github.com/adafruit/circuitpython/blob/1b7fbaf31eb84ce718fc715dc09a32b010eac9b5/ports/esp32s2/common-hal/microcontroller/__init__.c#L70-L73

I don't think it is possible to enter `DFU` (first-stage) bootloader on esp32s2 from software.
I tried to get reset to `UF2` bootloader working but I am facing a strange issue #3813.

ladyada, 2020-12-27:
there is some way to enter the DFU bootloader from usercode because arduino does it... 
@me-no-dev do you have any hints we could use to implement? :)

me-no-dev, 2020-12-28:
@ladyada here is the shutdown handler that does the reboot into the proper mode: https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/cores/esp32/esp32-hal-tinyusb.c#L458-L479

it is registered here: https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/cores/esp32/esp32-hal-tinyusb.c#L531

Let me know if this does not help :)

microDev1, 2020-12-28:
Thanks! @me-no-dev, this works... the changes are in my [`update-run-mode`](https://github.com/microDev1/circuitpython/tree/update-run-mode) branch. :)

dhalbert, 2021-03-31:
> Thanks! @me-no-dev, this works... the changes are in my [`update-run-mode`](https://github.com/microDev1/circuitpython/tree/update-run-mode) branch. :)

@microDev1 Could you make your changes into a PR?

microDev1, 2021-04-01:
> @microDev1 Could you make your changes into a PR?

This is stuck due to `idf@v4.2`. In my implementation, the function calls used to achieve this are part of `idf@v4.3`.

ThomasAtBBTF, 2021-04-16:
I just want to point out, that I am very interested that this feature will work in the not to distance future,

dhalbert, 2021-07-19:
This should be fixed now, with the latest merge, and also with the latest TinyUF2. Update your TinyUF2 bootloader to at least 0.5.0, following the instructions here: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader

PR #4925 (2021-07-19): Add RunMode.UF2 and update esp32s2 run-modes
This PR closes #3884.

- Adds `microcontroller.RunMode.UF2`
- Adds `UF2`, `NORMAL` and `BOOTLOADER` run-modes to esp32s2
closed 2021-07-19 by dhalbert