adafruit/circuitpython #3886: ESP32S2 UART deinit() causes restart/reload safe mode, hard fault handler on restart

rmincher, 2020-12-27:
Running 6.1.0-beta.2-182-g1b7fbaf31 on 2020-12-26; FeatherS2 with ESP32S2.

From REPL the following causes restart:
```
>>> from board import *
>>> import busio
>>> x = busio.UART(TX,RX,baudrate=115200)
>>> x.deinit()
```

UnexpectedMaker, 2020-12-28:
In 6.11beta2 I get the following error when trying to initialise the UART
```
>>> from board import *
>>> import busio
>>> x = busio.UART(TX,RX,baudrate=115200)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: IO44 in use
```

I can't even get to x.deinit()

microDev1, 2020-12-28:
I was able to replicate the crash...

```
Adafruit CircuitPython 6.1.0-beta.2-182-g1b7fbaf31 on 2020-12-26; microS2 with ESP32S2
```

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400b5347  PS      : 0x00060030  A0      : 0x800b4349  A1      : 0x3ffdccf0
A2      : 0x00000000  A3      : 0x3f02c1e0  A4      : 0x3ffcf564  A5      : 0x3ffcf560
A6      : 0x00000010  A7      : 0x00000000  A8      : 0x800b534c  A9      : 0x3ffdccd0
A10     : 0x00000001  A11     : 0x3f4c0048  A12     : 0x3ffcf48c  A13     : 0x0e7f1fd8
A14     : 0x00000003  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x3ffcf48c  LEND    : 0x0e7f1fd8  LCOUNT  : 0x40029479

Backtrace:0x400b5344:0x3ffdccf0 0x400b4346:0x3ffdcd10 0x400afb15:0x3ffdcd30 0x400928ba:0x3ffdcd50 0x4008e30d:0x3ffdcd70 0x4008e3fd:0x3ffdcd90 0x4009bede:0x3ffdcdb0 0x400929d3:0x3ffdce50 0x4008e30d:0x3ffdce80 0x4008e33a:0x3ffdcea0 0x400acc25:0x3ffdcec0 0x400acee9:0x3ffdcf70 0x4009f530:0x3ffdcfb0 0x4009f7c0:0x3ffdcfd0 0x4009fa43:0x3ffdcff0 0x400304c0:0x3ffdd010

ELF file SHA256: a434075c943f96b1

CPU halted.
```

> In 6.1beta2 I get the following error when trying to initialize the UART

This was fixed in #3817. Any **non** `DEBUG=1` build after commit https://github.com/adafruit/circuitpython/commit/d6ee17c6fd36cb5589d487d3f3d257a53bc805e3 will not have that error when using default uart pins.

microDev1, 2020-12-28:
Got a fix for this. Getting a PR ready.

UnexpectedMaker, 2020-12-28:
Thanks @microDev1 👍 
Any idea what the:
```ValueError: IO44 in use``` 
error is?

anecdata, 2020-12-28:
@UnexpectedMaker
> This was fixed in #3817. Any non DEBUG=1 build after commit d6ee17c will not have that error when using default uart pins.

UnexpectedMaker, 2020-12-28:
> @UnexpectedMaker
> 
> > This was fixed in #3817. Any non DEBUG=1 build after commit [d6ee17c](https://github.com/adafruit/circuitpython/commit/d6ee17c6fd36cb5589d487d3f3d257a53bc805e3) will not have that error when using default uart pins.

But is that specific to UART pins? I see this error with other pins as well. This was reported with my LDO2 pin when trying too turn on my second LDO in my shipping code.py. I just re-flashed 300 boards removing that line to stop the error/fail.

```Adafruit CircuitPython 6.1.0-beta.2 on 2020-12-03; FeatherS2 with ESP32S2
>>> import code
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code.py", line 10, in <module>
  File "feathers2.py", line 23, in enable_LDO2
ValueError: LDO2 in use
>>> 
```

So I'm just trying to clarify if this is a general fix or a UART specific one, as the original issue was UART specific.

microDev1, 2020-12-29:
Hi @UnexpectedMaker, the `ValueError` you are getting seems to be coming from your python code. 
The reason you are getting it now is because the way pins are claimed was changed in `6.1beta2` as it was flawed before.

closed 2020-12-29 by jepler