adafruit/circuitpython #3887: OSError on nRF52840 when accessing I2c-BNO080

yugesh-recursive, 2020-12-28:
**I get following error when accessing the SparkFun VR IMU Breakout - BNO080 (Qwiic) using I2C from  SparkFun Pro nRF52840**

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "adafruit_bno08x/i2c.py", line 27, in __init__
  File "adafruit_bno08x/__init__.py", line 506, in __init__
  File "adafruit_bno08x/__init__.py", line 512, in initialize
  File "adafruit_bno08x/__init__.py", line 1054, in soft_reset
  File "adafruit_bno08x/i2c.py", line 42, in _send_packet
  File "adafruit_bno08x/i2c.py", line 42, in _send_packet
  File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 19] Unsupported operation
```
**TEST CODE**

```
import time
import board
import busio
import adafruit_bno08x
from adafruit_bno08x.i2c import BNO08X_I2C

i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)
bno = BNO08X_I2C(i2c)
```

tannewt, 2020-12-28:
It's not clear to me that this is a CircuitPython core issue.

Please work this through with SparkFun support first. Their support forum is here: https://forum.sparkfun.com/

closed 2020-12-28 by tannewt

paulmand3l, 2021-02-27:
@yugesh-recursive did you ever figure out what this was? I'm running into the same error with a different sensor.

dhalbert, 2021-02-27:
@paulmand3l What sensor are you having trouble with? This kind of error can be due to I2C protocol issues. I am trying to track down a similar error on a different board.

paulmand3l, 2021-02-27:
Garmin LIDAR-Lite V4. Here's my forum post about it: https://forums.adafruit.com/viewtopic.php?f=57&t=176040

dhalbert, 2021-02-27:
I answered in the forum post.