adafruit/circuitpython #3889: Unexpected Maker FeatherS2 - SPI creation error

lwerner-lshigh, 2020-12-28:
When creating SPI on the default pins on the latest 6.1.0 beta 2 I receive a value error `ValueError: IO36 in use`. This happens both when I create the SPI myself `spi = busio.SPI(board.IO36, MOSI=board.MOSI, MISO=board.MISO)` I get the same error as well. But when I create this bus by hand in the 6.0.0 release it works fine. 

Sample 1:
```python
import board
spi = board.SPI()
```

Sample 2:
```python
import board, busio
spi = busio.SPI(board.IO36, MOSI=board.MOSI, MISO=board.MISO)
```

tannewt, 2020-12-28:
Please try the absolute newest build off of S3. We've had some pin reset fixes recently.

lwerner-lshigh, 2020-12-28:
Just to be sure the `f9d9c03` build?

lwerner-lshigh, 2020-12-28:
Cool can confirm that `f9d9c03` works!

closed 2020-12-28 by lwerner-lshigh