adafruit/circuitpython #3890: irremote corrupts auto-reload, and other things on Circuit Playground Express

robertgallup, 2020-12-28:
Using CircuitPython 6.0.0 on a Circuit Playground Express, after running the irremote-simpletest.py (below), auto-reload doesn't work (i.e. writing to CIRCUITPY doesn't force reload). Normal execution of a simple print loop with `time.sleep()` also doesn't seem to work properly. It hangs until keyboard input(?). The only recovery seems to be to re-install the bootloader AND CircuitPython (reset, and re-installing CircuitPython alone doesn't seem to fix the problem)

```
# Circuit Playground Express Demo Code
# Adjust the pulseio 'board.PIN' if using something else
import pulseio
import board
import adafruit_irremote

pulsein = pulseio.PulseIn(board.REMOTEIN, maxlen=120, idle_state=True)
decoder = adafruit_irremote.GenericDecode()

while True:
    pulses = decoder.read_pulses(pulsein)
    print("Heard", len(pulses), "Pulses:", pulses)
    try:
        code = decoder.decode_bits(pulses)
        print("Decoded:", code)
    except adafruit_irremote.IRNECRepeatException:  # unusual short code!
        print("NEC repeat!")
    except adafruit_irremote.IRDecodeException as e:  # failed to decode
        print("Failed to decode: ", e.args)

    print("----------------------------")
```

DavePutz, 2020-12-29:
I was unable to exactly reproduce this using CP 6.0.0 or 6.1.0-beta.2. I do see that while the irremote script is running auto-reload does not seem to work, but I had no issues with time.sleep or needing to reload the bootloader afterwards.

robertgallup, 2020-12-29:
Hmmm. So, I just duplicated the issue I'm seeing with CP 6.1.0-beta.2.

The exact procedure is:

1. Run `irremote_simpletest.py` by copying to `main.py` on the CPX
2. Make and save any change to `main.py` and note that auto-reload doesn't work
3. Use `CTRL-C` in serial console to stop the program.
4. Replace `main.py` with:

```
import time

while True:
	print ("Hello")
	time.sleep(.5)
```
5. Notice that auto-reload still doesn't work.
6. Use `CTRL-D` from the serial console to run the new `main.py`. Notice that no output appears until keystrokes are entered from the keyboard.

With CP 6.1.0-beta.2, I was able to restore normal behavior using hardware reset, so if you update `main.py` and use reset to run it, you would not notice a problem beyond `irremote_simpletest.py`.

tannewt, 2021-07-15:
@robertgallup Do you know if this is fixed in the latest 7.0.0 builds?

robertgallup, 2021-07-16:
No. It does not seem to be fixed. I just tried the irremote_simpletest.py example with CircuitPython 7.0.0-alpha.4 and and V7 adafruit-irremote.mpy library. Auto-reload still doesn't work. Also, remote_simpletest.py randomly exits after some passes through the While True loop, which is an error, I think.

PR #5100 (2021-08-12): Switch SAMD21 ticks to PER event
The EVSYS is used to generate an interrupt from the event. This
simplifies timing used in pulseio that conflicted with the
auto-reload countdown.

Fixes #3890
commit f0859ac9 (2021-08-12): Switch SAMD21 ticks to PER event

The EVSYS is used to generate an interrupt from the event. This
simplifies timing used in pulseio that conflicted with the
auto-reload countdown.

Fixes #3890
closed 2021-08-12 by tannewt