adafruit/circuitpython #3891: busdevice setup/init not working on 6.1.0 w/ CLUE nRF52840

siddacious, 2020-12-29:
Using the following test code and a SGP40 attached to the CLUEs Stemma QT port,
```
import board
import busio
import adafruit_bus_device.i2c_device as i2c_device

i2c = busio.I2C(board.SCL, board.SDA)
i2c_device = i2c_device.I2CDevice(i2c, 0x59)
print("i2c_device 'found'")
```
with **6.0.1** I get:
[image]

with **6.1.0 Beta 2** I get:
[image]

The only change here is flashing the new version

Interestingly, an I2C scan works as expected with 6.1.0 Beta 2:
[image]

dhalbert, 2020-12-29:
Maybe this is a dupe of #3795, fixed by #3847. But we are disabling native busdevice for now due to other issues: #3869.

You could test against the artifacts from #3847. For the CLUE, try https://adafruit-circuit-python.s3.amazonaws.com/bin/clue_nrf52840_express/en_US/adafruit-circuitpython-clue_nrf52840_express-en_US-20201218-fd21ff5.uf2. That is the #3847 merge commit, and is before #3869.

siddacious, 2020-12-29:
@dhalbert looks the same with the uf2 you linked to:
[image]
[image]

dhalbert, 2020-12-30:
Hmm, that's not what I expected. Could you also try the absolute latest from here: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/clue_nrf52840_express/en_US/

siddacious, 2021-01-04:
@dhalbert I tested the latest and it seems to work. 

I tested the previous (`d21ff5`) again as well and confirmed the failure.

Lmk what else you'd like me to try

dhalbert, 2021-01-04:
OK, great! So let's close this for now, and we'll make sure to retest when we get around to putting back native bus device.

closed 2021-01-04 by dhalbert