adafruit/circuitpython #3894: Feather S2 i2c issue - can't find board

djotaku, 2020-12-30:
I've got a Feather S2 and this sparkfun board: https://www.sparkfun.com/products/16294 . The sparkfun board is at address 0x60 vs the 0x67 that adafruit has for their MCP9600 CircuitPython library.

So I wrote this code:

```python
import time
import board
import busio
import adafruit_mcp9600
# frequency must be set for the MCP9600 to function.
# If you experience I/O errors, try changing the frequency.
i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
mcp = adafruit_mcp9600.MCP9600(i2c, address=0x60)  # sparkfun default address is 60
while True:
    print((mcp.ambient_temperature, mcp.temperature, mcp.delta_temperature))
    time.sleep(1)
```

I get this error:

 File "adafruit_bus_device/i2c_device.py", line 102, in write
OSError: [Errno 19] Unsupported operation

Then, after hitting control-D in the Mu REPL it just tells me there isn't an I2C device at 0x60.

Someone in the Adafruit Discord mentioned running this code: 

```python
"""CircuitPython Essentials I2C Scan example"""
# If you run this and it seems to hang, try manually unlocking
# your I2C bus from the REPL with
#  >>> import board
#  >>> board.I2C().unlock()

import time
import board

i2c = board.I2C()

while not i2c.try_lock():
    pass

try:
    while True:
        print("I2C addresses found:", [hex(device_address)
              for device_address in i2c.scan()])
        time.sleep(2)

finally:  # unlock the i2c bus when ctrl-c'ing out of the loop
    i2c.unlock()
```

When I did that, it showed up once and then it stopped discovering the board. 

On the Feather S2 I tried both with Circuit Python 6.0.1 and 6.1.0-beta2.

So, here are some steps I did to isolate the Feather S2:

- I tried different Stemma QT cables
- I tried the Adafruit AHT20 and that DID work with the Feather S2
- But for the sparkfun board that's giving me issues, I tried it with a QT Py and that worked perfectly, It worked both with my original code and with the I2C scan Adafruit code.

So, in conclusion: 

-  The Sparkfun board works with the QT Py (Circuit Python 6.0.0). 
- The Sparkfun board does not work with the Feather S2 neither with Circuit Python 6.0.1 nor Circuit Python 6.1.0-beta 2
- The Feather S2 isn't completely broken for I2C as it works with the AHT20 - or at least it pases the I2C scan for the AHT 20.

Thanks!

dhalbert, 2020-12-30:
We have been having trouble with I2C on the ESP32-S2. Could you try the "Absolute Newest" build for the Feather S2 from https://circuitpython.org/board/unexpectedmaker_feathers2/ . Third box down on the right side.

djotaku, 2020-12-30:
Will definitely give it a shot today. As another data point - I was able to get things to work using Arduino on that board.

djotaku, 2020-12-30:
using the UF2 for today (12/30) issue remains

djotaku, 2021-01-17:
loaded in today's build of Circuit Python for Feather S2 and with the same code running before ("""CircuitPython Essentials I2C Scan example"""). I'm now getting this error:

code.py output:
Traceback (most recent call last):
  File "code.py", line 18, in <module>
RuntimeError: SDA or SCL needs a pull up

Code done running.

dhalbert, 2021-01-17:
If you have no I2C breakouts (with pullups) connected, you will get that error. Do you have some I2C device connected?

djotaku, 2021-01-17:
Yes, the same board that led me to make the bug report - the sparkfun temp board.

djotaku, 2021-01-22:
Using today's UF2 for the S2 and yesterday's Circuit Python libraries seem to have undone the SDA/SCL pull up regression. The board now recognizes the Adafruit AHT20. Still hasn't solved the original issue. The sparkfun board that works with the Feather S2 in Arduino code still does not register in Circuit Python. In the I2C scan code above, it still shows [] for that board. It also seems to want to reboot a lot now.

djotaku, 2021-01-24:
# Section 1 - Trying to find the Sparkfun board in I2C scan

Threw on yesterday's bootloader onto the Feather S2. Still no luck.

BUT...

If I put the https://www.adafruit.com/product/3243 on top of the Feather S2 while the Sparkfun board is connected, then I got (this was via the Arduino serial monitor because if the Feather S2 would crash, it would unmount and that made it hard to debug with the Mu serial console):

16:57:43.479 -> I2C addresses found: ['0x60', '0x70']
16:57:46.594 -> I2C addresses found: ['0x60', '0x70']
16:57:49.709 -> I2C addresses found: ['0x60', '0x70']
16:57:52.823 -> I2C addresses found: ['0x60', '0x70']
16:57:55.938 -> I2C addresses found: ['0x60', '0x70']
16:57:59.053 -> I2C addresses found: ['0x60', '0x70']
16:58:02.167 -> I2C addresses found: ['0x60', '0x70']

Not sure what's up with that, but it makes it find the board. Still can't get temp readings, though because it complains about needing a pullup.

# Section 2 - Once again checking the AHT20 connected to the Feather S2 as a sanity check

I left the motor board on top since it was a pain to get in. I plugged in the AHT20 At any rate, with the following code:

```python

# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import board
import adafruit_ahtx0

# Create the sensor object using I2C
sensor = adafruit_ahtx0.AHTx0(board.I2C())

while True:
    print("\nTemperature: %0.1f C" % sensor.temperature)
    print("Humidity: %0.1f %%" % sensor.relative_humidity)
    time.sleep(2)

```

The Feather S2 runs just fine. 

# Section 3 - Using the Feather S2 with the Sparkfun board on the REPL

Trying to go manually with the Feature on REPL:

```python
>>> import board
>>> import time
>>> import busio
>>> import adafruit_mcp9600
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=10000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: SDA or SCL needs a pull up
```

# Section 4 - Sanity check on whether Sparkfun board truly needs pullups

I had the following code already running: 

```python

import time
import board
import busio
import adafruit_mcp9600

# frequency must be set for the MCP9600 to function.
# If you experience I/O errors, try changing the frequency.
i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
mcp = adafruit_mcp9600.MCP9600(i2c, address=0x60)  # sparkfun default address is 60

while True:
    print((mcp.ambient_temperature, mcp.temperature, mcp.delta_temperature))
    time.sleep(1)
    
``` 

Gives this output:
(22.125, 22.6875, 0.4375)
(22.125, 22.6875, 0.4375)

But wanted to go step by step to emulate what was happening on the Feather S2.

So here's the output of that:
```python
Adafruit CircuitPython 6.0.0 on 2020-11-16; Adafruit QT Py M0 with samd21e18
>>> 
>>> import time
>>> import board
>>> import busio
>>> import adafruit_mcp9600
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
>>> mcp = adafruit_mcp9600.MCP9600(i2c, address=0x60)
>>> print(mcp.ambient_temperature, mcp.temperature, mcp.delta_temperature)
22.625 24.1875 1.4375
```

ladyada, 2021-01-24:
whats the datecode on the MCP9600 chip

djotaku, 2021-01-25:
I looked at the chip under my magnifying glass and on there is printed:

MCP9600
E/MX (something in a circle I can't make out)
203182E

If that's not what you need and there's a way to get it in code, I can try to do that with the QTPy.

ladyada, 2021-01-25:
k yeah that date code is good, but the chip is generally super weird, it does not play well with i2c. i'd recommend going with another thermocouple amp.

djotaku, 2021-01-25:
Does that mean it's not likely to be some quirk of the ESP32-S2 that is yet to be sorted out? I guess it's not the end of the world if I have to do my project in Arduino code since it seems to work fine there. But it 's a bummer that it works in the PyQt in Circuit Python, but not here. 

Is there a K-type with the plug end (not bare wires), not MCP9600, and STEMMA QT that Adafruit sells that I just haven't seen in the store?

UnexpectedMaker, 2021-01-25:
No, it's is an ESP32-S2 thing... but not a quirk - just an "I2C needs more work on the CircuitPython side" of things. This is not the only outstanding issue with I2C on the ESP32-S2, but this specific issue is pretty niche, so @ladyada 's suggestion of using a different device is so your project is not stalled any longer.

djotaku, 2021-01-25:
Thanks for that reply - this is the first time I'm using a cutting edge board with either Adafruit or Arduino and so I wasn't sure if the suggestion to use another chip was an indication of where development was going. Please don't take my previous comment as whiny or ungrateful. I just read something between the lines that wasn't there. Thanks again, both the CPY team and @UnexpectedMaker  for your work on the board.

ladyada, 2021-01-25:
all good, the MCP9600 really just isnt that great. we're hoping the MCP9601 will fix that (but it isnt available to purchase yet)

cmumford, 2021-01-25:
> MCP9600 really just isnt that great

I ran into this a lot when using my [EZ Make Oven](https://learn.adafruit.com/ez-make-oven). It basically makes the oven unusable as the app hits this exception 100% of the time and kills the app. I hacked around it, but felt it was way too hacky to submit a PR for.

tannewt, 2021-01-25:
Does anyone know what the actual issue is? Has anyone taken a trace of the I2C to find the issue?

hierophect, 2021-02-15:
I'm consolidating this with a number of other issues that all appear to be the same thing. Please continue discussion of this issue in #4046.

closed 2021-02-15 by hierophect