adafruit/circuitpython #3897: Calling busio.UART.write from subclass hard-faults board. Does not enter safe-mode handler.

AdamCummick, 2020-12-30:
@jepler 
SAMD51 based board with build CircuitPython 6.0.0-278-g66fb09506

Uncomment ```super().write(b'hello')``` to cause the error.

```py

import board
import busio
import digitalio
import time
import random

class RS485(busio.UART):
    def __init__(self, tx: microcontroller.Pin, rx: microcontroller.Pin, de_pin, *, baudrate: int = 9600, bits: int = 8, parity: Optional[Parity] = None, stop: int = 1, timeout: float = 1, receiver_buffer_size: int = 64 , re_pin=None, auto_de_re=True, de_polarity=True, re_polarity=False, idle_transmitting=True):

        super().__init__(tx, rx, baudrate=baudrate, bits=bits, parity=parity, stop=stop, timeout=timeout, receiver_buffer_size=receiver_buffer_size)
        # super().write(b'hello') # running this hardfaults my board and does not enter the safe-mode handler
        print("Got here")
        #etc...

test = RS485(board.RX2, board.TX2, board.D6)

while True:
    pass

```

tannewt, 2020-12-30:
Why are you subclassing UART? Most native classes don't work with subclasses. We have to add code explicitly to the native class now to support it.

AdamCummick, 2020-12-30:
RS485 doesn't need too much more than some logic and timing around toggling a couple GPIO pins. Subclassing UART felt like the cleanest implementation, though I can avoid it if needed. I was unaware of the limitation of subclassing most native classes; if it is allowed, is that usually denoted in the documentation?

tannewt, 2020-12-30:
Ya, I think so. You could also add it to UART if you like. You can see the technique here in Group: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/displayio/Group.c#L79

dhalbert, 2020-12-30:
@AdamCummick said in discord:
> You mentioned rounding to the next power of 2 for the buffer, looking at the data sizes that could potentially add ~32kB of unneeded RAM. Would just raising an error to let the user make the call be better than that?

I was thinking more of buffer sizes in the tens or hundreds. Either is OK. In practice it seems people have always chosen a power of two, without thinking about it, or we would have heard about this before.

AdamCummick, 2020-12-30:
> Ya, I think so. You could also add it to UART if you like. You can see the technique here in Group: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/displayio/Group.c#L79

I tried adding this, but I'm still seeing the same crash: https://github.com/AdamCummick/circuitpython/commit/4809c9235468914231e9ee1a7e454a9e42166e96

Looking into the [origin of this on group](https://github.com/adafruit/circuitpython/commit/f29de5132562cd6e88661d150fa1ee3ab8b771e1) and #1881 it looks like it corrected for an object referencing a member before super().__init__(). This doesn't appear to be the problem with my code.

Is there something else that needs to be added or is my current implementation incorrect?

tannewt, 2021-01-05:
You need to have the functions call `native_uart` too. :-) Take a look at where `native_group` is called.

closed 2021-07-13 by tannewt