adafruit/circuitpython #3898: Implement DAC-based audio on ESP32-S2

dhalbert, 2020-12-30:
The DAC(s) support DMA, but it's not in the ESP-IDF API yet: https://github.com/espressif/esp-idf/issues/6333. So we would wait for that.

275RR, 2022-06-12:
any update on this? I have a a MagTag

ladyada, 2022-06-17:
i think we were seeing weird results, when @jepler is back we can follow up with espressif on whether they fixed DMA DAC support on the S2

gpeyton, 2022-10-11:
@jepler @ladyada any update on this?

ladyada, 2022-10-11:
@gpeyton since you are bumping - please check with espressif if they have DMA support for the ESP32-S2 DAC, that was our hold up as you can read above

tmkasun, 2024-10-09:
@ladyada and team, it looks like DAC DMA is now supported in Espressif: https://github.com/espressif/esp-idf/blob/master/components/esp_driver_dac/esp32s2/dac_dma.c.

If you could provide some guidance, I’d be happy to work on adding this support to CircuitPython.

tannewt, 2024-10-09:
@jepler Do you have a WIP branch somewhere that @tmkasun could start with?

Otherwise @tmkasun I'd suggest looking at [how I2SOut is implemented](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/audiobusio/I2SOut.c) and convert it into `audioio`.

jepler, 2024-10-09:
@tannewt no I don't think I have any WIP branch related to this

jepler, 2024-10-09:
The esp-idf docs make it sound like we would want to use continuous dma mode with asynchronous writing: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/dac.html#continuous-wave-output-continuous-dma-mode

the [on_convert_done callback](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/dac.html#_CPPv4N21dac_event_callbacks_t15on_convert_doneE) would be comparable to the function `i2s_event_interrupt` providing the next buffer to be filled in a background callback.

Thanks for looking at adding this to CircuitPython @tmkasun !

closed 2025-02-05 by dhalbert