adafruit/circuitpython #3906: Disallow subclassing a native class if it doesn't expect it

dhalbert, 2020-12-30:
It would be nice if CircuitPython threw an error if you tried to subclass a native class that isn't expecting it. Perhaps we could add a flag bit that it is allowed, with the default being that it is not.

Addresses issues like #3897.

tannewt, 2021-01-06:
It'd be even better if we could automatically get the superclass struct.

tannewt, 2021-01-06:
We can't automatically do it though because the native class may actually want the subclass. (Pixelbuf needs it for example.)

gneverov, 2023-02-06:
I think this is a responsibility for the individual classes to enforce themselves. In `make_new` classes should validate their `type` parameter and disallow the sub-classing as needed (by throwing `TypeError`).