adafruit/circuitpython #3920: ESP32S2 WiFi connect() "Connection Error: Unknown failure"

rpavlik, 2021-01-02:
Symptoms seem very similar to #3712 but I have no such dhcp weirdness on my network: I'm trying to connect it to my main AP/router. I have a MatrixPortal that connects fine, but the MagTag will either say network not found or "Connection Error: Unknown failure".

I'm using essentially the same test code as the other issue:

```python
import wifi
from secrets import secrets

for network in wifi.radio.start_scanning_networks():
    print("\t'%s'\t\tRSSI: %d\tChannel: %d" % (network.ssid, network.rssi, network.channel))
wifi.radio.stop_scanning_networks()
print("connecting...")
wifi.radio.connect(ssid=secrets['ssid'], password=secrets['password'])
print("connected.")
```

My network does appear there, but then we get:

```
connecting...
Traceback (most recent call last):
  File "code.py", line 8, in <module>
ConnectionError: Unknown failure
```

I've tested with 6.0 beta 3 as well as `adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-20210102-ad2ee09.uf2` - both act the same.

I have tried rebooting the router, no luck. (It's a TP-Link Archer C8: an N/AC router.)  It *can* connect to my IOT-specific router (a Linksys E2500 v3 N router running DD-WRT) OK but that router intentionally doesn't allow any of its clients to access the internet, so I can't run the samples that use internet-hosted APIs.

dhalbert, 2021-01-02:
Have you updated the firmware on the TP-Link Archer C8?

rpavlik, 2021-01-02:
Yes, it's as up-to-date as it can be (think it's version 1 hardware, so hasn't had updates in a few years). The weird thing is, the esp32 on the MatrixPortal works just fine, no problems. (and I have a swarm of esphome and tasmota esp8266 devices that connect to either, with no trouble.)

BennyE, 2021-01-02:
What is the security type please? WPA2-AES? TKIP?

rpavlik, 2021-01-02:
I'm pretty sure it's wpa2-aes, it's whatever the "good" one is. (Pretty sure my iot router is also using that though?)

BennyE, 2021-01-02:
Thank you! The router that "does not work" is using HT40 (channel width of 40 MHz) - your IOT router is set to "auto" which I'd assume would use 20 MHz. I'm not a fan of 40 MHz wide channels for 2.4 GHz, but this is not about my taste :). Can you test with 20 MHz wide channels, just to rule out that this is causing the issue? The comment about the space in the password could also be important - thank you -, I'd need to try this here tomorrow.

anecdata, 2021-01-03:
A space in the SSID _should_ be fine (I've tested that with other APs, and the specs allow it), but it's worth testing in this specific case. I'd also look at placement and orientation of the MagTag relative to the two routers.

rpavlik, 2021-01-03:
Thanks for the feedback. The second set of images are actually the working router, so the one that works is 40-wide (the esp devices all connect at 20 regardless, according to the ddwrt status screen), while the one that doesn't work is auto.

I can try going back to g/n on the main router: I think I set it for n-only for speed since I've got 300/300 internet. The two routers are about a foot apart in the same room (and the main/non-working one has much better range - I've gotten wifi in my neighbor's driveway). I've tried in a few rooms upstairs with both esp32 devices, no differences: matrix portal works everywhere, magtag fails everywhere.

Just to clarify, both SSID have no spaces, but the non-working network has spaces in the **password**

anecdata, 2021-01-03:
Sorry I missed that the space was in the password. You'd think you'd not be allowed to enter an invalid password, but definitely worth a try with no spaces.

rpavlik, 2021-01-03:
A space is definitely valid in general (I'm using that network to post this right now from my phone), but some devices appear to have software limitations. Will test with a spare router tomorrow: can't really change my family's main wifi when we're all here all the time, but I have another e2500 that I can set up and see if I can make the magtag fail to connect by putting in a password with spaces.

anecdata, 2021-01-03:
fwiw, I just created a WPA2 AP with a space in the password, and a 6.1.0-beta.2 esp32-s2 Saola was able to connect

rpavlik, 2021-01-04:
OK, so now I feel silly. It appears to work OK when brought down to my office (which is the next room over from the router), with no changes. Not sure why it appears to have shorter range than the matrix portal, but at least it works now. (The sample code says RSSI: -59 now when it connects.) Maybe the pick-and-placed standoff near the antenna is affecting it or something.

I think this can probably be closed now, since it seems like a signal thing probably.

closed 2021-01-06 by tannewt