adafruit/circuitpython #3922: UMFEATHERS2  no status LED

jerryneedell, 2021-01-03:
The Unexpected Maker FeatherS2 does not show the boot status via the onboard DotStar LED.
Was this intentionally omitted or should I look into implementing it?

jerryneedell, 2021-01-03:
perhaps relevant to this https://github.com/adafruit/circuitpython/commit/5c00ba8592fd0b9eba9e5c795630638a79fc4044

ladyada, 2021-01-03:
@unexpectedmaker please take a look! :)

UnexpectedMaker, 2021-01-03:
Thanks @ladyada :)
@jerryneedell When did you buy your FeatherS2?
The initial boards I shipped were before the UF2 bootloader was available for the ESP32-S2, so if you purchased when it was first released, your board may not have it on there. 
You can install it yourself though, I have instructions in my FAQ on https://feathers2.io

jerryneedell, 2021-01-03:
@UnexpectedMaker OK -- I did not realize the UF2 bootloader would impact the CircuitPython usage of the status LED. That is the blinking pattern that shows where an error occurs or the REPL status (Green/White)
I will try installing the USF2 bootloader and see if that changes things.

UnexpectedMaker, 2021-01-03:
@jerryneedell no, it has nothing to do with the stats LED outside of booting. You mentioned in your post it was not showing the boot status. Boot status is UF2. Error status during runtime is something else.

jerryneedell, 2021-01-03:
sorry for not being clear. I was really looking for the post boot REPL status and Failure status.

jerryneedell, 2021-01-03:
This is what I was hoping to be able to see https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/rgb_led_status.c#L114

UnexpectedMaker, 2021-01-03:
Ok, that's something different. I don't have the LED controlled by CPY as unlike other boards, the LED power can be disabled as it's connected to LDO2.

jerryneedell, 2021-01-03:
OK -- that is what I though -- and just wanted to understand better -- sorry for the confusion.
If LD02 is power by default after RESET, then I was hoping it could still be used for status until the user decided to turn it off.

UnexpectedMaker, 2021-01-03:
I can have a look at re-enabling it for CPY control for the next release. My initial tests of having this enabled was problematic, but that was back in early 6.0 alpha.

jerryneedell, 2021-01-03:
No rush and I'll be happy to take look at it myself.

UnexpectedMaker, 2021-01-03:
The 2 lines are already in ```mpconfigboard.h``` but are commented out. They were left there *in case* I ever wanted to re-enabled them. If you uncomment them and it works reliably for you, let me know, and I'll make it a permanent change :)

jerryneedell, 2021-01-03:
I tried that and it still did not work as I expected -- I was just going to look to see if there were some other flags that needed to be set.

jerryneedell, 2021-01-03:
looks like I also have to enable CP_RGB_STATUS_LED

jerryneedell, 2021-01-03:
No - I think I am confused about that -- that is for discrete RGB LEDs....  still looking

jerryneedell, 2021-01-03:
I think something like this is needed https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/microcontroller/Pin.c#L132 and https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/microcontroller/Pin.c#L96
I'll experiment with it and let you know how it goes.

jerryneedell, 2021-01-04:
@UnexpectedMaker I think I now have the status LED working -- I updated the common-hal/microcontroller/Pin.c and .h files to handle the pins  based on how it was done for the Atmel boards -- I now get the expected response from the dotstar.
I do some more testing to make sure the changes do not impact other builds and will submit a PR when I have it tested.

jerryneedell, 2021-01-04:
ah -- wait a minute -- the LDO2 in use is not my fault...  it also happens with the current main!

jerryneedell, 2021-01-04:
It looks like the LDO2 in use issue is just that you have to enable LDO2 before initializing the dotstar.  
It works fine if It do it that order. 

it appears that on powerup, LDO2 is enabled that is , it is High but it is set as an Input
```
Adafruit CircuitPython 6.1.0-beta.3-30-gec0210240 on 2021-01-04; FeatherS2 with ESP32S2
>>> import board,digitalio
>>> ldo2 = digitalio.DigitalInOut(board.LDO2)
>>> ldo2.value
True
>>> ldo2.direction
digitalio.Direction.INPUT
>>> 
```
Is there any problem using the DotStar with board.LDO2 set an an Input? 

I am also finding that even if I turn off LDO2 -- the Dotstar still responds ... is that expected?

jerryneedell, 2021-01-04:
@UnexpectedMaker PR #3930 submitted -- I was not able to ad you as as reviewer so hopefully a tag here is sufficient.
Let me know if I missed anything of if you want any changes.

UnexpectedMaker, 2021-01-04:
> It looks like the LDO2 in use issue is just that you have to enable LDO2 before initializing the dotstar.
> It works fine if It do it that order.

There was a change in CPY in 6.1 beta 1 or 2 that altered how IO objects where held on to, to bring the S2 port in line with other ports, which triggered this error in my code. I have an update set of code available here:
https://feathers2.io/code/shipping_files.zip

> 
> it appears that on powerup, LDO2 is enabled that is , it is High but it is set as an Input

> ```
> Adafruit CircuitPython 6.1.0-beta.3-30-gec0210240 on 2021-01-04; FeatherS2 with ESP32S2
> >>> import board,digitalio
> >>> ldo2 = digitalio.DigitalInOut(board.LDO2)
> >>> ldo2.value
> True
> >>> ldo2.direction
> digitalio.Direction.INPUT
> >>> 
> ```
> 
> Is there any problem using the DotStar with board.LDO2 set an an Input?

LDO2 is an output. Where is it set as an input? It controls a logic gate. If left floating, the logic gate falls back to VDDSDIO control, but should not be left floating.

> 
> I am also finding that even if I turn off LDO2 -- the Dotstar still responds ... is that expected?

Yes, unfortunately, the APA102's suffer from parasitic power via the CLK/DATA pins, but you'll notice it's much dimmer without LDO2 on. The APA needs to be set to clear when not in use, or it will still drain power from LDO1.

I've though about moving away from the APA and use something else, but the colour from the APA is much nicer than the WS/SK LEDs and it's PWMable.

jerryneedell, 2021-01-04:
My mistake -- It may well be an output in the core code but when you set it up via CP
ldo2=digitalio.DigitalInOut(board.LDO2)
it is set as in INPUT
So I guess that is not a problem

UnexpectedMaker, 2021-01-04:
If you are seeing DIM colour, then LDO2 is not enabled, so O21 is either not HIGH or is not an OUTPUT, so it's high impedance.

jerryneedell, 2021-01-12:
Addressed by #3930

closed 2021-01-12 by jerryneedell