adafruit/circuitpython #3924: MagTag Infrequent Failures

dcarmean, 2021-01-03:
I'm having problems with infrequent crashes on the MagTag that are similar to the problems I have with the Matrixportal (described [here](https://forums.adafruit.com/viewtopic.php?f=60&t=172275&p=844238#p844238)).  I've tried this with several versions of Circuit Python, and now have the board running 6.1.0-beta.3 with libraries from *-bundle-6.x-mpy-20210101.

The code is pulling data from the Purple Air and Weather Underground web sites.  It will run for a while (frequently hours) and then crash in a fetch_data() call.  I have a try/except clause around the fetch_data(), but for some reason it only catches the error occasionally.  

The code is attached the post on the forum [here](https://forums.adafruit.com/viewtopic.php?f=60&t=173673).

Here is the error:

```
> Retrieving data...Traceback (most recent call last):
>   File "code.py", line 126, in <module>
>   File "code.py", line 122, in <module>
>   File "adafruit_portalbase/network.py", line 453, in fetch_data
>   File "adafruit_portalbase/network.py", line 425, in fetch
>   File "adafruit_requests.py", line 612, in get
>   File "adafruit_requests.py", line 576, in request
>   File "adafruit_requests.py", line 436, in _get_socket
> OSError: 0
```

This is coming from this section of the code (full code attached):

           print("***inside")
           fetch_success = False

           try:
               inside = NETWORK.fetch_data(
                   url=Inside_SOURCE,
                   json_path=(PM2_Location,TEMP_Location,HUMI_Location),)
               fetch_success = True
           except RuntimeError as e:
               print("Some error occurred, retrying! -", e)
               continue

I have seen the try/except actually catch the error on some iterations.  For example, this will pop up every now and then:

```

> ***inside
> Retrieving data...Some error occurred, retrying! - Sending request failed
> Free Memory: 1,946,496
> Iteration:  13

```

I'll also file this as an issue as @siddacious suggested in the original post.

tannewt, 2021-01-06:
There are lots of errors that can happen with networking. Have you tried catching OSError explicitly?

dcarmean, 2021-01-06:
Yes, I modified the original code to include catching the OSError and reporting.  Once the OSError gets triggered, it looks like the network connection is gone and does not recover.  Here is the code snippet

`          print("***inside")
           fetch_success = False

           try:
               inside = NETWORK.fetch_data(
                   url=Inside_SOURCE,
                   json_path=(PM2_Location,TEMP_Location,HUMI_Location),)
               fetch_success = True
           except (RuntimeError, OSError) as e:
               print("Error occurred, retrying! -")
               print("\t",type(e))
               print("\t",e.args)
               print("\t",e)
               continue
`
Which produces the following output:

> Free Memory: 1,946,528
> Iteration: 33
> Time: 2021 - 1 - 4 8 : 57
> ***inside
> Retrieving data...Error occurred, retrying! -
> <class 'OSError'>
> (0,)
> 0
> Free Memory: 1,946,528
> Iteration: 34
> Time: 2021 - 1 - 4 8 : 58
> ***inside
> Retrieving data...Error occurred, retrying! -
> <class 'OSError'>
> (0,)
> 0
> Free Memory: 1,946,528
> Iteration: 35
> Time: 2021 - 1 - 4 8 : 59
> ***inside
> Retrieving data...Error occurred, retrying! -
> <class 'OSError'>
> (0,)
> 0
>

cmezak, 2021-01-15:
I was hitting this 100% when trying to connect to an octoprint server on my network. Turned out that replacing "octopi.local" with the IP address of the server fixed it completely! Hopefully this is at least a clue for the transient case @dcarmean is hitting.

tannewt, 2021-07-15:
We may not fix these errors but we should make the errno non-zero.

PR #5036 (2021-07-23): Remove OSError(0) and old network modules
The newer modules are `socketpool` and `ssl`.

Fixes #3924, related to #2021, closes #1880, closes #2273, closes #2274
commit f84cb948 (2021-07-23): Remove OSError(0) and old network modules

The newer modules are `socketpool` and `ssl`.

Fixes #3924, related to #2021, closes #1880, closes #2273, closes #2274
closed 2021-07-23 by microDev1