adafruit/circuitpython #3926: Double wake from Touch Alarm in fake sleep (ESP32S2)

dglaude, 2021-01-04:
I tested Touch Alarm with the ESP32S2 with the code sample in #3850 

When in fake sleep (USB connected to computer), it seems that my board wakeup twice in a raw when I touch.

The double trigger on touch only seems to occur with the USB connected to computer, but not when on a power supply.

Basically it made the code blink the blue LED twice in a raw for 5 seconds (with a single touch of the pin).
Tested with 6.1.0-beta3.

The wake-up speed is also much faster in fake sleep.

microDev1, 2021-01-04:
I have a vague memory of this issue being present in my early testing... 
However this issue was not present in my final testing of `TouchAlarm` PR and I am unable to replicate it now.

Tested with :-
```
Adafruit CircuitPython 6.1.0-beta.3 on 2020-12-31; microS2 with ESP32S2
``` 

Console output :- _(on a fresh-start)_
```
WakeAlarm :-  None

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.
# -------------------------> TouchAlarm was triggered here.
Woken up by alarm.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
WakeAlarm :-  <TouchAlarm>

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.
# -------------------------> Waits for TouchAlarm to trigger.
```

> The wake-up speed is also much faster in fake sleep.

I don't think this is specific to `TouchAlarm`... an image validation check is done when coming out of deep sleep which might be the cause of the delay here. This check can be skipped by editing the `sdkconfig`. More on that is [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/kconfig.html#config-bootloader-skip-validate-in-deep-sleep).

dglaude, 2021-01-04:
Here is the result of my test (consistent from what I remember from this week-end):

```
# -----------------------------------------> Doing CTRL+D
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
WakeAlarm :  None
Reason: None

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.
# -----------------------------------------> Touching the I01 pin => I get to wake up in a raw.
Woken up by alarm.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
WakeAlarm :  <TouchAlarm>
Reason: TouchAlarm

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.
Woken up by alarm.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
WakeAlarm :  <TouchAlarm>
Reason: TouchAlarm

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.
# -----------------------------------------> Just waiting => Alarm
Woken up by alarm.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
WakeAlarm :  <TimeAlarm>
Reason: TimeAlarm

Code done running.
Pretending to deep sleep until alarm, CTRL-C or file write.

```

Here is the code I use (that mix two alarm kind and Pin can be added):
```
"""
Testing TouchAlarm and TimeAlarm simultaneously on ESP32S2
"""

import time
import board
import alarm
import digitalio

# Recovering the reason of the wakeup
reason = alarm.wake_alarm
print("WakeAlarm : ", reason)

# Testing reason value to distinguish different kind of alarm
if reason is None:
    print("Reason: None")
elif isinstance(reason, alarm.time.TimeAlarm):
    print("Reason: TimeAlarm")
elif isinstance(reason, alarm.touch.TouchAlarm):
    print("Reason: TouchAlarm")
elif isinstance(reason, alarm.pin.PinAlarm):
    print("Reason: PinAlarm")
else:
    print("Reason: Unknown")

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
led.value = True

time.sleep(5)

# Create an alarm that will trigger 60 seconds from now.
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 60)
# Create an alarm that will trigger by the touch of a pin.
touch_alarm = alarm.touch.TouchAlarm(board.IO1)

#pin_alarm = alarm.pin.PinAlarm(board.IO0)

####Traceback (most recent call last):
####  File "code.py", line 37, in <module>
####TypeError: 'pin' argument required

# Exit the program, and then deep sleep until time or touch alarm wakes us.
alarm.exit_and_deep_sleep_until_alarms(time_alarm, touch_alarm)
print("This place will never be reached.")
```

microDev1, 2021-01-13:
@dglaude I tried your code but still can't replicate the issue. 
I made a few tweaks in `TouchAlarm` recently. Can you do a test with current `main`.

microDev1, 2021-03-25:
Closing due to lack of activity...

closed 2021-03-25 by microDev1