adafruit/circuitpython #3928: Request: Indicate which file failed to load

JetForMe, 2021-01-04:
When working with CP (especially on macOS), it’s likely one will get `Corrupt .mpy file` messages. It would be immensely helpful if these messages also indicated which file it was talking about.

I took a quick look at the code, and in `py/persistentcode.c`:

```c
STATIC void raise_corrupt_mpy(void) {
    mp_raise_RuntimeError(translate("Corrupt .mpy file"));
}
```

There’s also:

```c
mp_raw_code_t *mp_raw_code_load_file(const char *filename) {
    mp_reader_t reader;
    mp_reader_new_file(&reader, filename);
    return mp_raw_code_load(&reader);
}
```

One way to implement this would be to save `filename` in a static file-global variable, and then reference it when constructing the error strings (see the handful of `translate` calls throughout the file). That will take up a pointer’s worth of RAM, though, so it might be better to pass the filename around through all the calls, which is more tedious, but not terrible.

I don’t know what other concerns there might be, and I don’t have time at the moment to get set up to build CP (I really gotta get this other thing done), but I believe in a general philosophy of error messages actually being helpful. I figure this hint should make it trivial for someone steeped in the code to implement, or perhaps indicate why it’s a bad idea.

Thanks.

tannewt, 2021-01-06:
I think this is a good idea overall. I wouldn't do it with a global though.

This is the first I've heard of regularly getting corrupt mpy though. That worries me more. Are you regularly overwriting the mpy?

JetForMe, 2021-01-06:
I agree I wouldn't do it with a global, either. Really, the loader should be a class that allocates some space for the duration of the load, and puts stuff like the filename there.

Anyway, no, I was running into issues copying `.mpy` files to the `lib` folder on a Feather M0 Basic using the Finder on macOS, due to it copying `._` versions of the files (not sure why it does that or why they're so big). I think there may also be a free space reporting issue. In any case, `cp -Xr` worked to solve that issue, but in the meantime I was managing to partially-copy `.mpy` files. Had I known *which* file it didn’t like, I would’ve figured this out sooner 😝