adafruit/circuitpython #3937: Scroll back in output on MatrixPortal

ajs256, 2021-01-06:
On the MatrixPortal, it is very hard to see output since the display is so low-resolution. I would like to be able to scroll in the output with the UP and DOWN buttons, so I can see why my project crashed and not have to watch the NeoPixel. ("wait for a dark gap... there! Wait, was that orange or yellow? And which line was it on? Ok, starting over...")

This could probably also be applied to other boards with displays and buttons, such as the PyBadge family.

FoamyGuy, 2021-01-06:
I always end up plugging in to USB and watching the output in the Serial Console instead of trying to decode the LED colors.

But I also really like the idea of having scrolling in the built-in displayio terminal output.

ajs256, 2021-01-06:
I suppose another way to address this problem (a project crashes, how do I see why it crashed?) could be to store exception data somewhere that persists across resets, so I can plug the board into my computer and check it from the REPL.

tannewt, 2021-01-07:
@cwalther was experimenting with saving the string after the VM finished.