adafruit/circuitpython #3947: ESP32-S2 add support for WPA3.

caiokat, 2021-01-07:
I would like to use WPA3 with my FeatherS2. This should probably be an optional feature since WPA3 uses more resources than WPA2. What do you guys think?

https://github.com/caiokat/circuitpython/commit/d4671011efb5dea686dffb2ba551721f998a9d22

tannewt, 2021-01-07:
I think we should always build the ability into the firmware. It's good to have consistency.

Does the `pmf_cfg` setting prevent WPA2 from working or does it automatically choose?

@bennyE Wants this too.

caiokat, 2021-01-07:
`pmf_cfg.capable = true;` shouldn't prevent WPA2 from working. There's another setting `pmf_cfg.required`, if this is set the ESP32-S2 will only connect to AP if AP supports Protected Management Frames.

I disabled WPA3 on my AP and the FeatherS2 automatically connected to it using WPA2 instead.

BennyE, 2021-01-07:
Hi @caiokat & @tannewt, great to find like-minded people on such topics - really happy! When I tested this directly with the ESP-IDF some time ago, it was necessary to set both capable and required to successfully connect to a WPA3(SAE-AES)-only SSID. IEEE 802.11w which is Protected Management Frames (PMF) aka Management Frame Protection (MFP) is not new with WPA3, but it was made mandatory to support by equipment that wants to claim WPA3 compliance. You could find WPA2 SSIDs that require you to do PMF, which is great against deauth-attacks.

There is an "easy" mode which is Transition-Mode (or backward-compatible) SSID that allows both WPA2 and WPA3, which look like this in the air:
[image]

As per my experience/knowledge (which may have changed in the meantime) the WPA3-only SSID requires to have this setting on ESP32 (tested with Huzzah at that time):
```c
            .pmf_cfg = {
                .capable = true,
                .required = true
            },
```
[image]

The challenge I saw with this, is that the [wifi_ap_record_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/network/esp_wifi.html#_CPPv416wifi_ap_record_t) doesn't tell you if the AP/SSID requires PMF or is just capable. We could safely assume that for WPA3_SAE, but a "discovery" would be better.

caiokat, 2021-01-08:
Hi @BennyE. Unfortunately my AP only supports transition mode so I can't test that. But I would love to have WPA3-only support, too.

We could start with transition mode first since it's so easy to implement and it doesn't seem to break anything.

```c
.pmf_cfg = {
    .capable = true,
    .required = false
}
```

tannewt, 2021-01-08:
@caiokat That sounds like a good start to me too.