adafruit/circuitpython #3948: Is there a plan to support esp32-s3?

kingsjl, 2021-01-07:
Is there a plan to support esp32-s3?

ladyada, 2021-01-07:
yes, once we get chips we can investigate adding support. there's no eta and we don't even have samples yet :)

closed 2021-01-07 by ladyada

street-grease-coder, 2021-01-31:
@ladyada they shipped out samples to some makers on youtube, ahead of world wide release time

ladyada, 2021-01-31:
@street-grease-coder hi what are the specific links for folks who have em?

street-grease-coder, 2021-01-31:
@ladyada  can I pm you somehow?

ladyada, 2021-01-31:
nope

street-grease-coder, 2021-01-31:
https://www.youtube.com/watch?v=tcrFiZuUi18
https://www.youtube.com/channel/UCu7_D0o48KbfhpEohoP7YSQ
I would include more details, but not here, hope that's useful to you anyways..

ladyada, 2021-01-31:
hi - that video is esp32-c3 not s3?

StuartIanNaylor, 2021-02-03:
The esp32-s3 (S3) could be amazingly useful from KWS to image recognition but if Espressif try to shoehorn into Espressif only solutions of esp32-who & esp32-skainet its likely going to be a flop.
If we get anywhere near a complete port of tensorflow-lite then wow its got big potential.

ladyada, 2021-10-12:
hi, sub to https://github.com/adafruit/circuitpython/issues/4363 for status updates