adafruit/circuitpython #3950: How do you enable WEBUSB in CircuitPython?

kbanks-krobotics, 2021-01-07:
I have seen the capability demo'd in youtube video, so I know it is possible. What I can't find is where/how you enable it.
I can tell that supervisor/supervisor.mk launches a python program tools/gen_usb_descriptor.py.
What I can't tell is the intended way to add in an additional descriptor such that WEBUSB is included in the composite devices.

The code seems to support "CDC" "MSC" "HID" "AUDIO" (and possibly "MIDI"). Is there a "WEBUSB"? Is there supposed to be?

Does gen_usb_descriptor.py as-written support WEBUSB? and it is just a matter of passing in the correct command-line parameters from supervisor.mk?

Or does gen_usb_descriptor.py need more work, plus new parameters need to be provided from supervisor,mk?

Or am I looking in the wrong place completely?

To ask another way - how was https://www.youtube.com/watch?v=5UU8GRoDw8c done IN CIRCUITPYTHON? (where were the code changes?)

ladyada, 2021-01-07:
that demo was written in arduino not circuitpython

kbanks-krobotics, 2021-01-08:
Has anyone done similar work in CircuitPython? (just trying not to reinvent a wheel here...)

ladyada, 2021-01-08:
they have not! you're welcome to PR webusb support, we simply did not get to it

kbanks-krobotics, 2021-01-08:
I will see what I can do. High-level plan is to simply have WEBUSB enabled TOO, and you will be able to user EITHER serial interface with the REPL. That said, I'm still learning about these "composite" USB devices and they may not be as usable "in parallel" as I am thinking.

FiriaCTO, 2021-02-04:
The functionality of the tinyusb "WebUSB serial" demo has been incorporated into a webusb branch. It turns out you **can** have your CDC console and a WebUSB console at the same time, but depending on how many USB endpoints your chip supports, you might have to give up SOMETHING. For some ESP32-S2 work we are doing at Firia Labs, we have been trading USB HID for WebUSB at compile time. A PR has been submitted.

dhalbert, 2021-02-04:
Note that another alternative is the "Serial SPI", sometimes known as "WebSerial". This was first mentioned in closed issue #605.

Spec: https://web.dev/serial/
Status in Chrome: https://www.chromestatus.com/feature/6577673212002304

The [Chrome demo](https://googlechromelabs.github.io/serial-terminal/) does not quite work with CircuitPython (I see repeated `<DISCONNECTED> <CONNECTED>`). But I think we could get it to work

FiriaCTO, 2021-02-05:
PR has been merged into main - THANKS dhalbert!
For those interested in this feature, please take a look at WEBUSB_README.md

ladyada, 2021-02-05:
great! thanks @firiacto!

closed 2021-02-05 by ladyada