adafruit/circuitpython #3952: How does light sleep interact with background tasks?

jepler, 2021-01-08:
Working on I2S for esp32s2 (#3733), I discovered that doing a time.sleep() is either
 * EITHER causing the `i2s_event_task` not to run
 * OR causing the work it registers with `background_callback_add` not to be performed
To compensate for this, I've included code to enable background ticks whenever an I2SOut object exists.  However, my expectation had been that "interrupt-like" activities that create background callbacks would still occur, and those background callbacks would be executed, during light sleep.

If this makes sense, and we can change it so that this is the case, great -- otherwise, we can improve our internal documentation so that we don't collectively forget it.

tannewt, 2021-01-08:
FreeRTOS is weird compared to our other ports. Sleep is done by waiting on task notification: https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/supervisor/port.c#L293

Anything that wants to wake the main task needs to notify it. See https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/supervisor/port.c#L293

So, if your i2s code happens outside the main task, then you need to notify the cp task to have it run anything.

If notifications occur outside of the cp task's wait, it'll cause the next wait to return immediately.

jepler, 2021-01-08:
So in `background_callback_add_core()` we need to call this port-specific function so that an ongoing sleep will be interrupted.  On most arches it would do nothing, but on esp32s2 it would call `xTaskNotifyGive(circuitpython_task);`.  @tannewt suggests `port_wake_main_task()`.

closed 2021-01-12 by dhalbert