adafruit/circuitpython #3961: Builds failing due to tinyusb

TG-Techie, 2021-01-10:
## Description 

When trying to build a fresh clone of main (cloned 2021Jan09 after 22h00) the nrf port failed to build (see the issue below).
From some investigation, it seems that the tinyusb dependency that is pulled during the  `git submodule` update has either a bug in it or relies on overflow for behavior. 

`@Mikew67#1266` on discord also experienced this issue. 

### Testing:
I have some old clones of circuitpython (6.0-alpha range) which still build. After manually copying the old `lib/tinyusb` into the fresh clone and building again the below error did not occur. (A different error occurred, I suspect it is due to breaking changes in tinyusb's api)

### Versions/build info:
- the builds of the fresh clone were on main and 6.1.x
- the feather_nrf52840_express was used in each case to test if the build succeeded  (also test TG-Watch02A)
- gmake == 4.3
- arm-none-eabi-gcc == 7.3.1 

Solutions:
- None yet. (check the comments below incase one is found)

## The Error:
```
CC ../../lib/tinyusb/src/common/tusb_fifo.c
../../lib/tinyusb/src/common/tusb_fifo.c: In function 'tu_fifo_write':
../../lib/tinyusb/src/common/tusb_fifo.c:143:6: error: assuming signed overflow does not occur when assuming that (X + c) < X is always false
 [-Werror=strict-overflow]
   if ((p > p + offset) || (p + offset > f->max_pointer_idx))
      ^
cc1: all warnings being treated as errors
gmake: *** [../../py/mkrules.mk:55: build-feather_nrf52840_express/lib/tinyusb/src/common/tusb_fifo.o] Error 1
```

#### Secondary Error from Testing:
```
CC ../../supervisor/shared/workflow.c
../../supervisor/shared/workflow.c: In function 'supervisor_workflow_connecting':
../../supervisor/shared/workflow.c:39:12: error: implicit declaration of function 'tud_connected'; did you mean 'tud_connect'? [-Werror=impli
cit-function-declaration]
     return tud_connected() && !tud_suspended();
            ^~~~~~~~~~~~~
            tud_connect
```

TG-Techie, 2021-01-10:
## Solution Found
 
I was able to find a hackey one and a proper solution. 
In short, my gcc compiler was out of date.

### Cause:
After discussion on discord, `@joey#0903 `  pointed out that they have also had this issue in the past and were directed to use a newer version of the gcc compiler, "2019-q4 release".
 See Joey's discord conversation [here](https://discord.com/channels/327254708534116352/327298996332658690/797712197379293214)

### Testing:
after installing arm-none-eabi-gcc 9.2.1 the build works with clean, "un-hacked" tinyusb code. (tested running on a board)

## Proposed Fixes:
- Add a check to make files to warn developer if they are using a gcc version that is older than what is suggested
- Add a section to the readme specifying the suggesting compiler version.
- (not suggested) turning off overflow checking or that file.

### Hackey Solution:
Instead of deleting this "solution", I'm keeping it here to show that there is another possible solution if it is neede.
Okay, after some looking around in the tinyusb library it seems like `src/common/tusb_fifo.c` is the only file that relies on overflow for bound checking of fifo pointers.

when editing the pulled submodule in `src/common/tusb_fifo.c` there are two changes that allow the file to compile:
by storing the result of an overflow check in an intermediate variable the compiler does not raise a warning.

### Hacks:
#### original for advance_pointer: (src/common/tusb_fifo.c line 143)
```
  if ((p > p + offset) || (p + offset > f->max_pointer_idx))
```
#### Hackey fix: 
```
uint16_t p_plus_offset = p + offset;
  if ((p > p_plus_offset) || (p_plus_offset > f->max_pointer_idx))
```

#### original for babackward_pointerkw: (src/common/tusb_fifo.c line 159)
```
  if ((p < p - offset) || (p - offset > f->max_pointer_idx))
```
#### Hackey fix: 
```
 uint16_t p_minus_offset = p - offset;
  if ((p < p_minus_offset) || (p_minus_offset > f->max_pointer_idx))
```

dhalbert, 2021-01-11:
We switched to the `10-2020-q4-major` toolchain as of recently, so that's the best one to use:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

`Unable to check whether maximum number of endpoints is respected` is expected: it's a compile-time check that we can't always do.

dhalbert, 2021-01-11:
Btw, you think tinyusb should do something differently, it's best to file an issue there: https://github.com/hathach/tinyusb.

closed 2021-01-11 by dhalbert