adafruit/circuitpython #3962: analogio refuse to change the value

dglaude, 2021-01-10:
Testing on
> Adafruit CircuitPython 6.1.0-rc.0 on 2021-01-06; Adafruit Feather M0 Express with samd21g18

analogio seems to fail when I try to change the value:
```
import board
from analogio import AnalogOut
analog_out = AnalogOut(board.A0)
analog_out.value = 1000
analog_out.value = 2000

```

The result is and error on the line that set the value to 2000:

> main.py output:
> Traceback (most recent call last):
>   File "main.py", line 5, in <module>
> ValueError: Object has been deinitialized and can no longer be used. Create a new object.

When trying the guide code from https://learn.adafruit.com/circuitpython-essentials/circuitpython-analog-out it fail on the line: `analog_out.value = i`

dhalbert, 2021-01-10:
Reproduced on SAMD21, but SAMD51 is OK.

dhalbert, 2021-01-10:
@jepler I bisected this to https://github.com/adafruit/circuitpython/pull/3541/commits/fb768dfc14ab3b98905927f984a080e551513ea8

What I see when debugging is that the `AnalogOut` object is getting smashed between the first and second call to `common_hal_analogio_analogout_set_value()`. It seems to be OK immediately after the `dac_sync_write()`.

OK:
```
$2 = {base = {type = 0x35cc0 <analogio_analogout_type>}, descriptor = {device = {hw = 0x42004800}, sel_ch = {{
        buffer = 0x0, length = 0}}}, channel = 1 '\001', deinited = false}
```
later:
```
$4 = {base = {type = 0x35cc0 <analogio_analogout_type>}, descriptor = {device = {hw = 0x42004800}, sel_ch = {{
        buffer = 0x0, length = 0}}}, channel = 224 '\340', deinited = 125}
```

dhalbert, 2021-01-11:
Channel number is getting set wrong; will fix.

PR #3966 (2021-01-12): fix atmel-samd DAC channel selection logic
Fixes #3962 (@dglaude).

- Pin channel `if` statements tested opposite condition.
- `PA05` check not guarded by SAMx5 check.

Rewrote as a `switch` statement.

Tested on SAMD21 and SAMD51.
closed 2021-01-12 by dhalbert