adafruit/circuitpython #3968: Deep sleep current is up to 2.6mA on ESP32-S2

dhalbert, 2021-01-11:
After the latest round of deep sleep additions, deep sleep current is up to about 2.6mA on the MagTag. It was about 260uA before. Let's try to get it back down.

Confirming what @UnexpectedMaker was seeing on the FeatherS2 boards with the latest builds.

dhalbert, 2021-01-11:
This is due to setting up touch-alarm support. If I comment out the call to `alarm_touch_touchalarm_prepare_for_deep_sleep()` in `alarm_enter_deep_sleep()`, it goes back down to 260uA. Will see about not enabling touch if `TouchAlarm` is not in use.

dhalbert, 2021-01-11:
@microDev1 Do you have time in the near future to work on this? If not we will try it. Should not be hard: `PinAlarm` keeps a count of how many pin alarms are going to be used and does nothing if none are being used for a deep sleep.

UnexpectedMaker, 2021-01-11:
> This is due to setting up touch-alarm support. If I comment out the call to `alarm_touch_touchalarm_prepare_for_deep_sleep()` in `alarm_enter_deep_sleep()`, it goes back down to 260uA. Will see about not enabling touch if `TouchAlarm` is not in use.

Ok, so that begs the question: Why so high when RTCGPIO is enabled? That still way outside the max os 260uA that Espressif state on their data sheet. I know my board overhead is < 30uA so Max I should be seeing is approx 300uA with worst deepsleep level.

dhalbert, 2021-01-11:
`touch_pad_set_meas_time()` sets the duty cycle for touch measurements, but it is not called in the CPy code. I don't know what the deafults are.

UnexpectedMaker, 2021-01-11:
@dhalbert What version of CPY was the last that was working properly pre PinAlarm?

microDev1, 2021-01-11:
> This is due to setting up touch-alarm support. If I comment out the call to `alarm_touch_touchalarm_prepare_for_deep_sleep()` in `alarm_enter_deep_sleep()`, it goes back down to 260uA. Will see about not enabling touch if `TouchAlarm` is not in use.

Yeah... I suspected that.

> @microDev1 Do you have time in the near future to work on this?

Sure. :-)
- In the PR above I fixed _"touch alarm initialization if not set"_ so it can be in `6.1.0`.
- Current optimization can be done as an enhancement.

`esp_sleep_pd_config()` is explicitly called with `ESP_PD_DOMAIN_RTC_PERIPH`... auto pd config doesn't work. 
This turns on RTC IO, sensors and ULP co-processor.

dhalbert, 2021-01-11:
> @dhalbert What version of CPY was the last that was working properly pre PinAlarm?

I tested a version I happened to have on a MagTag that did not have `PinAlarm` or `TouchAlarm`. 6.1.0-beta.2 did not have either. Both were added in 6.1.0-beta.3.

dhalbert, 2021-01-11:
@tannewt with just a `PinAlarm` enabled on D11 with `pull=True`, I am seeing 1.7mA during deep sleep. Is this what you saw? It is higher than I would expect.

tannewt, 2021-01-11:
> @tannewt with just a `PinAlarm` enabled on D11 with `pull=True`, I am seeing 1.7mA during deep sleep. Is this what you saw? It is higher than I would expect.

Yup, that's about what I was seeing.

UnexpectedMaker, 2021-01-12:
@dhalbert Ok, running beta 2 I can confirm I am getting around 85uA now with CPY - which matches my Arduino values.

dhalbert, 2021-01-16:
Closing this because the values are expected. We can work on lowering sleep currents, but the regression is fixed for now.

closed 2021-01-16 by dhalbert