adafruit/circuitpython #3971: Better document what AnalogIn's reference_voltage means

jepler, 2021-01-11:
The `reference_voltage` of an AnalogIn is the value that lets you scale from the 0..65535 integer value to a floating-point voltage value.  However, the documentation also states that the ADC can read the whole range from 0V to 3.3V (for example)

On the esp32-s2, the reference_voltage is 3.3, and doing `adc.value * adc.reference_voltage / 65535` gives the accurate analog voltage but only in a range of about 0.1 to 2.6V.

```
//|     reference_voltage: Optional[float]
//|     """The maximum voltage measurable (also known as the reference voltage) as a
//|     `float` in Volts."""
```
so this docstring should be updated, but I didn't have a good enough text to suggest updating it to.

kevinjwalters, 2021-01-22:
Is there a reason for `reference_voltage` to be optional?

ladyada, 2021-01-22:
yes, you can also change the aref on some chips either internally or externally. many chips have default tat folks dont use

tekktrik, 2022-01-14:
How about:

```
reference_voltage: Optional[float]
//|     """The maximum voltage measurable (also known as the reference voltage) as a
//|     `float` in Volts.  Note the ADC value may not scale to the actual voltage linearly
//|     at ends of the analog range."""
```

tekktrik, 2022-01-14:
Happy that (or something else) to add it to the documentation!

closed 2022-02-09 by dhalbert