adafruit/circuitpython #3975: _bleio: Unable to write more than 20 bytes to a peripheral characteristic when acting as central

jlopez, 2021-01-12:
I'm using the nRF52840 as a central and trying to write to a BLE peripheral's characteristic. The data written is 26 bytes long. But it fails with:

```
ValueError: Value length > max_length
```

The `Characteristic.max_length` seems to be set to 20 bytes, the default. And there's no way to specify a different one as the characteristic is created internally by `Connection.discover_remote_services`

The device I'm trying to connect to is a Spark 40 by Positive Grid. Here's the code I'm using:

```python
from _bleio import adapter, UUID

PREFIXES = b"\x03\x03\xc0\xff"
SVC_UUID = UUID(0xFFC0)
CHR_UUID = UUID(0xFFC2)
DATA = b"\x01\xfe\x00\x00\x53\xfe\x1a\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf0\x01\x00\x00\x00\x02\0x23\xf7"

def find_chr(svcs, uuid):
    return next(c for s in svcs for c in s.characteristics if c.uuid == uuid)

for scan_entry in adapter.start_scan(prefixes=PREFIXES):
    adapter.stop_scan()
    conn = adapter.connect(scan_entry.address, timeout=1)
    print("Found")

if conn:
    print("Connected")
    services = conn.discover_remote_services((SVC_UUID,))
    characteristic = find_chr(services, CHR_UUID)
    if characteristic:
        print("Writing value")
        # Will throw ValueError: Value length > max_length
        characteristic.value = DATA
    conn.disconnect()
```

The output:
```
Found
Connected
Writing value
Traceback (most recent call last):
  File "code.py", line 23, in <module>
ValueError: Value length > max_length
```

dhalbert, 2021-01-13:
@jlopez Could you try this test UF2, which I hope will fix your issue? It's for a Feather nRF52840. If you need it for another board, just ask. You will need to unzip it, because GitHub won't allow uploads of .uf2's directly.

[feather840-6.1.0-rc.0-146-gccace62ac.uf2.zip](https://github.com/adafruit/circuitpython/files/5810976/feather840-6.1.0-rc.0-146-gccace62ac.uf2.zip)

jlopez, 2021-01-13:
Sure thing. I have an ItsyBitsy nRF52840, though. Once I get it I'll give it a shot. Thanks!

dhalbert, 2021-01-13:
@jlopez 

[itsy840-6.1.0-rc.0-146-gccace62ac.uf2.zip](https://github.com/adafruit/circuitpython/files/5811453/itsy840-6.1.0-rc.0-146-gccace62ac.uf2.zip)

jlopez, 2021-01-13:
It works! The write now succeeds without an exception. Thanks again! 😄

dhalbert, 2021-01-13:
Great. WIll submit a PR shortly.

PR #3993 (2021-01-14): don't check length for remote characteristic or descriptor
Fixes #3975. Fix tested by @jlopez, who file the original issue.

`_bleio`: When doing a remote a remote characteristic or descriptor write, the size of the remote attribute is not known. The code was mistakenly checking against the default values in the characteristic or descriptor proxy object. This was a actually a regression: the original code had the check in the right place. Moved the checks so they only happens when writing a local value.
closed 2021-01-14 by tannewt