adafruit/circuitpython #3988: Resetting either programmatically or via watchdog causes device to go into SAFE_MODE

dheera, 2021-01-13:
Running this:
```
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.NORMAL)
microcontroller.reset()
```
causes the board to reset and go into safe mode instead of normal mode.

Similarly, setting up a watchdog with the following code.py:
```
from microcontroller import watchdog as w
from watchdog import WatchDogMode
w.timeout = 10.0
w.mode = WatchDogMode.RESET
w.feed()
while True:
    pass
```
causes the microcontroller to reset after 10.0 seconds (as expected), but the microcontroller resets into safe mode instead of resetting and re-running code.py, which is the usual intent of watchdog usage.

Product: FeatherS2
CircuitPython version: adafruit-circuitpython-unexpectedmaker_feathers2-en_US-6.1.0-beta.3.uf2

Related but not same issue: https://github.com/adafruit/circuitpython/issues/3884

microDev1, 2021-01-13:
There are two different issues here :-
1. `microcontroller.on_next_reset()` :- Only `RunMode.SAFE_MODE` is currently implemented.
2.  The watchdog reset gets caught here :-
    https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/supervisor/port.c#L141-L151

microDev1, 2021-01-13:
There are two different issues here :-
1. `microcontroller.on_next_reset()` :- Only `RunMode.SAFE_MODE` is currently implemented.
2.  The watchdog reset gets caught here :-
    https://github.com/adafruit/circuitpython/blob/92a439899624145f3597dcaaabb84880c30e5185/ports/esp32s2/supervisor/port.c#L142-L151

dhalbert, 2021-05-22:
`microcontroller.reset()` works for me if I use the .bin, but not if TinyUF2 is present.

dhalbert, 2021-07-15:
The reason is stored in a supposedly "reserved by ESP-IDF". Check if TinyUF2 is using the same RTC register as CircuitPython. Try a different one?

dheera, 2021-07-16:
I believe this was already resolved by this pull request I made that was accepted
https://github.com/adafruit/circuitpython/pull/4011

dhalbert, 2021-07-16:
> I believe this was already resolved by this pull request I made that was accepted

Thanks! We think there may still be a problem with UF2 builds and will re-test.

closed 2021-07-19 by dhalbert

ThinkTransit, 2021-08-10:
Hi,

I have tested this on the nightly build from last night - Adafruit CircuitPython 7.0.0-alpha.5-175-g8da98eb1f on 2021-08-10; FeatherS2 with ESP32S2

microcontroller.reset() works as expected however resetting from the watchdog still goes to safe mode on the FeatherS2

FlorianBaumgartner, 2021-10-02:
Yes, same problem as @ThinkTransit described. My setup: Adafruit ESP32-S2 Metro, UF2 (0.5.2), cpy (7.0.0).
It's impossible to boot into the UF2-Bootloader from user code. Either by using circuit python:
`import microcontroller`
`microcontroller.on_next_reset(microcontroller.RunMode.NORMAL)`
`microcontroller.reset()`

or directly in Arduino CPP:
`esp_reset_reason();`
`esp_reset_reason_set_hint((esp_reset_reason_t) 0x11F2);`
`esp_restart();`

It acts just as a regular reset (single button press). By reading the reset reason I always get: "RTCWDT_RTC_RESET"