adafruit/circuitpython #3994: Separate TLS implementation into ssl.SSLSocket

hierophect, 2021-01-14:
Currently, the Socketpool/Socket API somewhat awkwardly crams two different libraries into the same implementation, using LWIP for non-encrypted TCP and datagrams, and the IDF/mbed TLS library for encrypted communication. Since TLS isn't really a transport layer library, it would be better to separate it into a different implementation of Socket, so the current Socket code doesn't need to infer what kind of socket it's dealing with for each call.

This is a follow up issue to PR #3854.

hierophect, 2021-01-14:
@tannewt this would be implemented all the way through shared-bindings, correct? So we'd have a new Socket.c file under `ssl` recreating the functions that can use TLS?

tannewt, 2021-01-14:
Yes. It should be `SSLSocket` to match CPython: https://docs.python.org/3/library/ssl.html?highlight=ssl#ssl.SSLSocket

hierophect, 2021-02-04:
Resolved by #4049

closed 2021-02-04 by hierophect