adafruit/circuitpython #3995: Adding new methods to Socket

hierophect, 2021-01-14:
This issue is for tracking or suggesting new methods and features for Socket. There are a number of methods not yet implemented for the Socket module, such as:

```
getsockopt()
setsockopt()
recv()
sendall()
getblocking()
setblocking()
getpeername()
getsockname()
```

Some methods may not be appropriate for Circuitpython, and others may be low priority but also trivial to implement. If you have a specific feature you require from Socket to support a library or project, please mention it here.

tannewt, 2021-01-14:
We shouldn't do any of them that allocate memory. (I think `recv` does.) It's a better practice to hand memory in.

hierophect, 2021-01-15:
@tannewt Makes sense. Do you think it would be worthwhile to actually throw not-implemented exceptions for them? Or a message that very briefly explains why Circuitpython avoids those calls compared to normal CPython?

tannewt, 2021-01-15:
@hierophect No, let's just omit them. Throwing errors for everything we omit would be too much. (We leave lots of `os` and `time` out already.)

hierophect, 2021-01-15:
that's reasonable, I only mention it because I've seen `recv` used so much in example tutorials for Socket online - but if we have our own tutorials for getting servers set up and such, it could be mentioned there as a footnote.