adafruit/circuitpython #3998: [Feature Request] Support SparkFun RV-8803

BiatuAutMiahn, 2021-01-14:
Can you support the SparkFun RV-8803?

Product Page: https://www.sparkfun.com/products/16281

Thank you

jepler, 2021-01-14:
Agreed -- this looks like a neat chip and it would be awesome to be able to add it to the Community bundle!  Check out https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library/overview to see how that process goes.

tannewt, 2021-01-14:
@BiatuAutMiahn Make sure and let SparkFun know you'd like a CircuitPython library for it too.

BiatuAutMiahn, 2021-01-15:
@tannewt, Done. I sent them an email

node-, 2021-03-24:
For what it's worth, I crudely adapted the Arduino interface for use with a Raspberry pi. Useful in the interim.

https://github.com/node-/Python-SparkFun-RV8803-RTC