adafruit/circuitpython #3999: Support ESP32-NOW

hierophect, 2021-01-14:
[ESP32-NOW](https://www.espressif.com/en/products/software/esp-now/overview) is a custom connectionless protocol developed by Espressif, which allows ESP32 and ESP32S2 boards to communicate directly with each other without using WiFi. It's supposedly similar to the 2.4GHz connectivity used in wireless mouses. This would enable ESP32S2 projects to connect multiple devices without the need for a router or any kind of internet connection, such as gathering data from a network of remote sensors, talking between a remote control and a receiver, sending instructions to a swarm of receiver units, etc. In terms of the projects it enables, it'd be similar to a lower range, built in LORA radio.

[This link](https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/) seems to have some good examples (all written for Arduino). Espressif documentation is [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html).

Some of the traits of ESP32-NOW include:
 - It's connectionless, so no handshaking
 - Open field range reportedly about 200m
 - Works simultaneously with wifi (though it's a bit finicky)
 - Short packets, 250 bytes max per packet.
 - 10 peers max when encrypted, up to 20 unencrypted

askpatrickw, 2021-01-14:
Question: Would this be a library or must it be implemented in the ESP32-S2 port?

hierophect, 2021-01-14:
@askpatrickw it will need to be implemented on the port level, it doesn't make use of existing wifi calls, if that's what you mean. But we might want a library on top of it.

microDev1, 2021-01-18:
How about a new `MESH` module for this... the top-level stuff can also be shared with Zigbee (#4007).

anecdata, 2021-01-18:
There's also ESP-MESH which is a more typical wi-fi mesh protocol. Naming may get confusing.

microDev1, 2021-01-21:
> How about a new MESH module for this... the top-level stuff can also be shared with Zigbee (#4007).

I got confused between `ESP-NOW` and `ESP-MESH` here. 
`ESP-NOW` seems to be more of a point-to-point receiver-to-transmitter communication rather than a mesh network.

ehagerty, 2021-01-21:
FWIW, i'm using their esp-mdf C stack right now doing some PoC for my healthech startup - utilising their meshkit-sense, meshkit-light and soon meshkit-button.  Their reference code for the button actually uses both esp-mesh and esp-now modes.  Don't know if that is useful to you guys at all.

microDev1, 2021-01-21:
> Their reference code for the button actually uses both esp-mesh and esp-now modes.

Interesting... `ESP-NOW` is more lightweight than `ESP-MESH` so one can assume it would work better when it comes to sensor or button nodes which are not always active and require power consumption to be minimal.

BetterAutomations, 2021-05-07:
Since this is being integrated into [MicroPy](https://github.com/micropython/micropython-esp32/issues/197) and [Pycom](https://forum.pycom.io/topic/7016/new-beta-release-v1-20-3-b3?page=1) could their modules be imported in CircuitPy?

hierophect, 2021-05-07:
> Since this is being integrated into [MicroPy](https://github.com/micropython/micropython/pull/6515) and [Pycom](https://forum.pycom.io/topic/7016/new-beta-release-v1-20-3-b3?page=1) could their modules be imported in CircuitPy?

It'll certainly make implementation easier. I'm thinking of tackling this for a personal project so I'll check it out over the weekend.

hierophect, 2021-05-11:
For anyone who is interested in this module - what kinds of projects would you use it for? And what do you feel the biggest draws of it are, compared to other similar options?

BetterAutomations, 2021-05-16:
> For anyone who is interested in this module - what kinds of projects would you use it for? And what do you feel the biggest draws of it are, compared to other similar options?

I would use it for accessories connecting to a central unit. The biggest drawback for me is the requirement that the central and the accessory knows each other’s MAC address. When selling accessories I cannot know this in advance. The MAC must be shared on another channel first such as Bluetooth. Another drawback is that the protocol isn’t open; only ESPs can talk to ESPs by ESP-NOW.

vkichline, 2021-09-02:
ESP-NOW is a great remote control protocol. It can achieve **much** longer range (with **much** lower bandwidth) than WiFi and needs no network access, making it resistant to many issues that affect portable control systems, such as security.
I would use this protocol today if it were available to write a wireless multi-servo control, using an M5 for the UI/HID and an ESP32-S2 as the receiver/controller.

SultanOfSolder, 2021-09-15:
> For anyone who is interested in this module - what kinds of projects would you use it for? And what do you feel the biggest draws of it are, compared to other similar options?

For community theater, props that can trigger their own sound queues. Think "How do we trigger a gun shot sound when the actor pulls the trigger, instead of having the sound guy inevitably triggering it 1.5 seconds late in at least one performance?" {Trust me, this happens all the time!}
The transmitter is in the prop, and the receiver sends a MIDI command when triggered, or signals a WAV file player that is plugged into the sound board.
The biggest draws are fast response, a private network that is less likely to be hacked, and lower power consumption.

tannewt, 2021-09-15:
@SultanOfSolder How far away is the prop from receiver? You could use BLE instead of Wifi.

wifijt, 2021-09-15:
BLE on the ESP32 is inconsistent and not compatible with the Nordic BLE -

SultanOfSolder, 2021-09-15:
@tannewt It varies, but can easily be 50 to 75 feet, line of sight. One of the theaters I work with is 110 seats, another is 1700. You could place the receiver above/below the stage and run a cable back to the sound board, but that's less than ideal. Not every show will need it, so I'm looking for a PnP solution.

deshipu, 2021-09-15:
I would also love to have the ability to use ESP-Now for my robots. Bluetooth is not an option, since this is an ESP32-S2. I will use regular WiFi for now.

BetterAutomations, 2021-09-16:
Remember that other high speed protocol devices exist, such as LoRA or the HC-12 module. Cheap and effective, not secure but you can always add that on top in software.

SultanOfSolder, 2021-09-16:
@BetterAutomations All of which require an antenna of some length that you may not be able to bury in the prop. The arduino libs work, but it's a lot easier to figure out what you wrote in python when you have to make changes a year later.

grahamwarby, 2021-10-07:
Given the fact that esp now connects in milliseconds and wifi takes 6-7 seconds battery life for deep sleep etc. will be greatly extended. My application is for wands for my grandchildren that control leds.

BaunLib, 2021-11-24:
ESP NOW for offline communications in areas without internet?

tomasApo, 2021-11-24:
Not really, the bit rate is extremely small. An example of the use for this tech is to transfer temperature/humidity readings to another esp microcontroller with distances less than 200m. > ESP NOW for offline communications in areas without internet?

BaunLib, 2021-11-24:
Wishing to utilise the WIFI on Raspberry Pi or ESP32 for education delivery in rural communities

hierophect, 2021-12-09:
hoping to maybe have a crack at this over christmas

BaunLib, 2021-12-10:
Dear Sir
BAUN ROBOTICS Limited of Nigeria is configuring a school security solution
for offline remote locations in Nigeria.
Target schools are located in areas without* INTERNET,* Electricity and
patchy mobile cellular networks
Schools will be equipped with WIFI enabled switches to raise alarm, if
attacked, under siege or challenged, for help.
Solarpowered  Raspberry Pi 3 server based will manage the alert trigger
Alert via offline WIFI will be sent to preconfigured first responders
within 3 Kilometers from the schools.
There is provision of a fail safe facility to send alerts directly if local
management do not respond.
Herewith attached is our concept.
Please advise application of ESP32, ZIGBEE, etc and or the Range Extender
for this concept.
We await your news.
Femi Fadairo

ladyada, 2022-02-16:
heres a minimal implementation for anyone who wants to take a crack at adding to circuitpy https://github.com/chrmlinux/tinyESPNow

richsad, 2022-04-03:
@hierophect did you ever get ESP32-NOW working on CP? I have a number of ESP32-S2-based dev boards and I'd be interested in experimenting with this protocol for the same type of use cases where'd I'd use LORA or other lightweight radio system. The advantages of this protocol seem to be:

- No additional radio required for an ESP32
- better suited to low power use cases than WiFi
- does not require a router 
- ideal for small (<=250 byte) packets

If this is still ongoing perhaps I can help with the code. I'm a systems level programmer with lots of experience implementing libraries to access low level systems. I'd work on this as part of my hobby work with microcontrollers (primarily purchased from Adafruit)

deshipu, 2022-04-03:
> If this is still ongoing perhaps I can help with the code.
@richsad I don't think anybody is working on it at the moment, so if you feel like this, just go ahead and try it. There shouldn't be any problems merging it, as far as I know, and the devs were always happy to review such patches.

muttstang, 2022-05-02:
Something like this would be very useful for a project I'm working on currently.  I have 2 boards.  One in a stationary box on my boat and another in a hand controller.  I'd like to try to keep communication between the 2 boards as simple as possible

askpatrickw, 2022-05-02:
@muttstang  I've not heard anyone talk about it.

americoperez49, 2022-09-20:
> It'll certainly make implementation easier. I'm thinking of tackling this for a personal project so I'll check it out over the weekend.

Have you made any progress on this? If not, if you could give me some guidance on how to start, I would be willing to try to work on this. I desperately want this feature and considering that there are builds of micropython out there with ESP-NOW support, maybe it won't be super difficult to complete this.

askpatrickw, 2022-10-17:
@americoperez49  I think the first place to track is the MicroPython (MP) pull request.  https://github.com/micropython/micropython/pull/6515

That will need to be merged and released and then CircuitPython (CP) will need to pickup that new MP release before someone in the CP community can work on this.

askpatrickw, 2022-10-18:
Also related issue ... 
https://github.com/glenn20/micropython/issues/4

closed 2023-03-10 by dhalbert