adafruit/circuitpython #4000: Copyright notice on ReadTheDocs still shows 2014-2020

hugodahl, 2021-01-14:
I noticed that the copyright date on the [CircuitPython ReadTheDocs site](https://circuitpython.readthedocs.io/) is still displaying 2020. I realize that it doesn't (yet) show 2021 since there haven't been any updates for the site in 2021 yet, however, when the site is updated to support the next release of CP, it should display 2021.

Looking over the docs section of the code, and this history of the configuration line ([`/conf.py`](https://github.com/adafruit/circuitpython/blob/47015d778944d8f30d45c76446d4a1dc4ad66a45/conf.py#L106)), this appears to be something that is manually updated when noticed or opportune. It may be worthwhile having the "to" year of the copyright statement be determined programmatically when the docs are generated so that the statement reflects the correct time period.

tannewt, 2021-01-14:
@hugodahl Want to make a PR to update it?

hugodahl, 2021-01-14:
Can do!

commit fac4d9c5 (2021-01-15): Merge pull request #4006 from hugodahl/Update-documentation-copyright-date

Update documentation copyright date (Closes #4000)
closed 2021-01-15 by tannewt