adafruit/circuitpython #4007: NRF52: Support for Zigbee Protocol

maziarzamani, 2021-01-15:
I am currently working with the Zigbee protocol on nrf52840 in Zephyr and was actually thinking about switching to CircuitPython however I have not seen any Zigbee related API. Is anyone working on this as we speak?

tannewt, 2021-01-15:
I don't know of anyone! Let us know how we can help you add it.

maziarzamani, 2021-01-15:
> I don't know of anyone! Let us know how we can help you add it.

Where do we begin? :-) As I understand its an entire new protocol that has to be added to the framework the same way as **wifi**?

tannewt, 2021-01-20:
@maziarzamani The first thing to do is determine what the new API will be. The research phase of coming up with this is collecting links to other implementations and listing the uses we want to support. Once we have examples, I usually document the goal API and write non-working examples for it.

Bonus points for any existing Python zigbee APIs since we could choose to follow their API.

dhalbert, 2021-01-20:
Zigbee is a licensed protocol, though it appears that it could be used for free without certification. I have not seen a really clear answer about this. We are currently using the nRF SoftDevice for BLE. There is SoftDevice support for Zigbee, but again, I don't know if there are licensing restrictions. I would assume Zephyr has its own Zigbee stack. Are there restrictions on its use?

maziarzamani, 2021-01-20:
For me, it doesn't really matter whether its nrf52 or any of the Silicon Labs IC's which are currently utilized by IKEA for their Trådfri series. I am currently using both IC's for ZigBee applications, and at the end of the day I'd rather use SiLabs, but I am assuming no such porting really exists?

dglaude, 2021-01-20:
Did someone say Trådfri? Wake me up if there is a way to integrate CircuitPython with Trådfri... my house is slowly transforming to that technology.

maziarzamani, 2021-01-20:
Maybe the easiest way to get around this is to use the XBee3 modules, which already supports MicroPython.

maziarzamani, 2021-01-20:
> Did someone say Trådfri? Wake me up if there is a way to integrate CircuitPython with Trådfri... my house is slowly transforming to that technology.

Likewise, but I am not really looking to reverse engineer any "TRÅDFRI protocol" here, but merely adding a new one which works with a different system (Conbee etc.).

dhalbert, 2021-01-20:
Certainly, a CircuitPython library could be written for an external module, and there would be no licensing issues. That is probably much less work than supporting Zigbee via the native SoftDevice implementation or a transplanted zephyr implementation.

ryang14, 2021-01-20:
The STM32WB55 series runs the radio firmware on a second core. This lets it act like a discrete module integrated into the chip.

tannewt, 2021-01-20:
> I'd rather use SiLabs, but I am assuming no such porting really exists?

I've looked at their line up of chips but haven't found ones that provide something unique to CP. We can support anything with 32k RAM and native USB but there is a non-trivial amount of work to support new peripherals.

maziarzamani, 2021-01-20:
I am sort of considering pivoting to the [XBee3](https://www.digi.com/products/embedded-systems/digi-xbee/rf-modules/2-4-ghz-rf-modules/xbee3-zigbee-3) which already runs MicroPython out of the box and can be bought as dev boards or modules. All the (nasty) low-level ZigBee code abstraction is already dealt with. At this point I do not see the hassle in trying to reinvent the wheel on nrf52.

chbarg, 2021-02-04:
> 
> 
> I am sort of considering pivoting to the [XBee3](https://www.digi.com/products/embedded-systems/digi-xbee/rf-modules/2-4-ghz-rf-modules/xbee3-zigbee-3) which already runs MicroPython out of the box and can be bought as dev boards or modules. All the (nasty) low-level ZigBee code abstraction is already dealt with. At this point I do not see the point in all the hassle with trying to reinvent the wheel on nrf52.

As far as I know, the XBees do not have the Zigbee cluster communication fully implemented in Micropython.

nospam2000, 2021-05-09:
Yout might want to have a look at those links:
[MicroPython and Ikea HomeSmart](https://trmm.net/Ikea/)
Ikea HomeSmart product insights and how to hack into them. They are using the EFM32 chips

[Gecko Zigbee Python interface](https://github.com/osresearch/ZbPy)
This is a pure Python implementation of a IEEE802.15.4 and ZigBee network stack that works with MicroPython for the EFM32 chips

tbjers, 2022-01-05:
Howdy, I was directed to this issue when I asked about CP Zigbee/Z-wave, and was wondering if anybody here has made any headway into writing anything for CircuitPython? Or, if you have done research but not written anything and would like help implementing something. I would be starting from scratch, so any pointers would be helpful. But, seeing as I missed 24 pull requests in 2021, I might as well make up for it in 2022.

migs35323, 2024-11-15:
Heh, a bit off topic maybe;
I was hoping with adafruit's release of the esp32-c6 feather boards that we would finally get zigbee support in circuitpython
i feel like when you pre-order a game and get disapointed cuz hype lol, I got a couple of those boards hooping to work with zigbee out of the box, but still waiting lol