adafruit/circuitpython #4012: We need to revert PR#3930 please

UnexpectedMaker, 2021-01-17:
We need to revert the work done in this please:
https://github.com/adafruit/circuitpython/pull/3930

It completely prevents the user from shutting down LDO2 and any expected behaviour a user would expect after that happens.

CP is driving the APA all the time, so parasitic current is leaking form the APA and causing 1.4-1.6V to come out of the 3V3O pin when LDO2 is shutdown.

It's also confusing folks as the APA is now "doing it's own thing" all of the time now, not just when the user is in the REPL.

Sorry folks!
Seon

jerryneedell, 2021-01-17:
closed by #4013

closed 2021-01-17 by dhalbert

tannewt, 2021-01-20:
> CP is driving the APA all the time, so parasitic current is leaking form the APA and causing 1.4-1.6V to come out of the 3V3O pin when LDO2 is shutdown.

Is this a larger issue? We shouldn't communicate with the APA when LDO2 is off then right?

> It's also confusing folks as the APA is now "doing it's own thing" all of the time now, not just when the user is in the REPL.

The APA should be blinking like it would on all CircuitPython boards. It was a bug before when it wasn't being consistent. Note, we do want to change it's behavior with CP 7 so that it's more power friendly. Until then though, it should be consistent with other boards.

UnexpectedMaker, 2021-01-20:
> > CP is driving the APA all the time, so parasitic current is leaking form the APA and causing 1.4-1.6V to come out of the 3V3O pin when LDO2 is shutdown.
> 
> Is this a larger issue? We shouldn't communicate with the APA when LDO2 is off then right?
> 
Well, yes, I guess if it's considered a bug for the APA not to be behaving the same as other boards, then the way to solve this is  for the code that does the LED thing to only do it if LDO2 )IO21) is high,  but that's FeatherS2 specific.

The question the is, should LDO2 being forced as output/high on boot? I don't like that as that means the LDO is turned on without the user doing it, rather than by the users hand. If it's not forced one, then the LED won't do the "CP" thing, and would then still be considered a bug.

> The APA should be blinking like it would on all CircuitPython boards. It was a bug before when it wasn't being consistent. Note, we do want to change it's behavior with CP 7 so that it's more power [...]

See above re bug, and yay for better implementation in CP7 !

tannewt, 2021-01-20:
> The question then is, should LDO2 being forced as output/high on boot? I don't like that as that means the LDO is turned on without the user doing it, rather than by the users hand. If it's not forced on, then the LED won't do the "CP" thing, and would then still be considered a bug.

I need to think more about this. I expect more boards in the future to do similar. I like that CircuitPython could manage when the board is on automatically but understand the 1mA of neopixel power (or whatever) may not be desired. The benefit of automatically turning on would be using the RGB status on "power on reset" to indicate the safe mode wait (and I suspect a BLE wait for boards with it).