adafruit/circuitpython #4015: VS Code MagTag - ModuleNotFoundError

eukrit, 2021-01-17:
Hi AdaFruit/CircuitPython,

I'm new and I was trying to follow this instruction: 
https://learn.adafruit.com/magtag-covid-tracking-project-iot-display/circuitpython-internet-test

I've got a silly obstacle on VS Code related to the module not found as per the attached screenshot.
I think I've got all the files right as per the attached images.

Would you please help?

[image]
[image]
[image]
[image]

Extensions Installed: Arduino, CircuitPython, Python, Pylance
Latest Library: adafruit-circuitpython-bundle-6.x-mpy-20210117
Latest MagTag UF2: adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-6.1.0-rc.1

dhalbert, 2021-01-17:
That is VSCode not finding the module, but it doesn't mean it's not working on the board. Did you `pip3 install adafruit-circuitpython-requests` in Windows?

This kind of support issue is better handled in https://adafru.it/discord or https://forums.adafruit.com. Thanks.

closed 2021-01-17 by dhalbert

eukrit, 2021-01-17:
Thank you dhalbert.