adafruit/circuitpython #4016: Comment Fix - QSPI SCK and CS Reversed

mrangen, 2021-01-17:
I just noticed a comment that could be improved for clarity, as I referenced it and it led me to think I had my QSPI SCK and CS pins reversed.

circuitpython/ports/atmel-samd/boards/itsybitsy_m4_express/mpconfigboard.h

15    // DotStar pins, QSPI CS, and QSPI SCK
16    #define MICROPY_PORT_B (PORT_PB02 | PORT_PB03 | PORT_PB10 | PORT_PB11)

As it stands, PORT_PB02 and PORT_PB03 are understood to be the DotStar Pins, PORT_PB10 the QSPI CS, and PORT_PB11 the QSPI SCK. While it is correct about the DotStar pins, the QSPI CS and QSPI SCK are reversed.

I recommend replacing the comment on Line 15 with:

// DotStar SCK, DotStar MOSI, QSPI SCK, and QSPI CS

ladyada, 2021-01-17:
good eyes! could you submit a PR with the fix? :)

mrangen, 2021-01-18:
Sure thing! It'll be my first!

ladyada, 2021-01-18:
awwwseome!
here's a guide that may help
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

closed 2021-01-20 by tannewt