adafruit/circuitpython #4026: VID/PID Request for new board design

rfefx, 2021-01-18:
Hello. 
I am creating a derivative board for the Grand Central. I will be using the ATSAMD51P20A

Requesting a VID/PID for the new board. 

Thanks so much.

skerr92, 2021-01-18:
Hey there! If you’re needing a VID/PID checkout pidcodes. I believe @tannewt has been pushing this direction. He has access to the review process and can get it sorted there.

rfefx, 2021-01-19:
So i am following this guide on adafruit. https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/overview
which let me to this pull request. Pidcodes requires the open source software and hardware. I can't get my board added to circuitpython without the VID/PID so it's a catch 22.

skerr92, 2021-01-19:
Submit your pull for pidcodes and let Scott know what you’re doing. As long as the PID is available, you should have no issues pushing to CP for a PR. (I’ve done this twice already)

I am happy to help if need be.

tannewt, 2021-01-20:
> So i am following this guide on adafruit. https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/overview
> which let me to this pull request. Pidcodes requires the open source software and hardware. I can't get my board added to circuitpython without the VID/PID so it's a catch 22.

You can open source the code in your own fork to satisfy the pid.codes requirement and then PR to CircuitPython after.

dhalbert, 2021-03-31:
@tannewt described a path forward. Closing for now. Happy to help further if you are still stuck.

closed 2021-03-31 by dhalbert