adafruit/circuitpython #4027: Write to filesystem

iot49, 2021-01-19:
How can I boot the device such that the filesystem is available for writing from CircuitPython?
E.g. to log sensor readings.

Setup: device is connected by usb (and repl, for commands only). Macos.

`boot.py`:

```
storage.remount("/", False)
```

From `repl`:

```
with open('x.txt', 'w') as f:
    f.write(b'abcdefg')
```

Error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 30] Read-only filesystem
```

**Notes**:

- This used to work.
- Workaround: eject the device, then issue `storage.remount('/', readonly=False)` and write. Unfortunately e.g. when the (unattended) device is loosing power and reboots, it can no longer write to the filesystem.
- I know it's possible to disable auto-mount on a mac or linux, but I'd prefer not to.

dhalbert, 2021-01-19:
Did you `import storage` in boot.py? `boot_out.txt` will show output from running `boot.py`, including errors.

Another possibility is that the filesystem is corrupted, which would make it read-only even from CircuitPython.

I just tried this on a Metro M0 Express running the latest build from `main`, and it works for me.

iot49, 2021-01-26:
Yes, this works. Sorry for the false alarm.

closed 2021-01-26 by iot49

pianpie, 2022-08-09:
I have a similar issue to this. I imported storage correctly but the filesystem is stuck in in read-only for circuit python. I followed [this](https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage) guide and it didn't work. The filesystems mode was never changed. I am using the metro m4 express.
Boot.py:
```
import board
import digitalio
import storage

# For Gemma M0, Trinket M0, Metro M0/M4 Express, ItsyBitsy M0/M4 Express
#switch = digitalio.DigitalInOut(board.D2)

# For Feather M0/M4 Express
switch = digitalio.DigitalInOut(board.D2)

# For Circuit Playground Express, Circuit Playground Bluefruit
# switch = digitalio.DigitalInOut(board.D7)

switch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UPswitch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UP
# If the switch pin is connected to ground CircuitPython can write to the drive
storage.remount("/", switch.value)
```
code.py:
```
import time
import board
import digitalio
import microcontroller
import storage
led = digitalio.DigitalInOut(board.LED)
led.switch_to_output()
switch = digitalio.DigitalInOut(board.D2)
switch.direction = digitalio.Direction.INPUT
switch.pull = digitalio.Pull.UP
print(switch.value)
try:
    print("in try")
    with open("test.txt", "a") as fp:
        while True:
            fp.write("this works")
            led.value = not led.value
            print("pog")
            time.sleep(1)
except OSError as e:
    print("os error")
    delay = 0.5 
    if e.args[0] == 28:
        delay = 0.25
    while True:
        led.value = not led.value
        time.sleep(delay)
        print(switch.value)
```
The filesystem isn't being changed and it's stuck in read-only for circuit python

EricAlm, 2023-03-14:
Hi, how do I send a newline when writing to the file?
already tried

fp.write("\x0A")
fp.write("\n")
fp.write("\r")

and none worked for me