adafruit/circuitpython #4029: PWM on Trinket M0

obiben, 2021-01-20:
Hardware: Trinket M0
Bootloader: 3.10.0
CircuitPython: 6.0.1

I'm having trouble using PWM on Trinket M0.

Examples point me towards a library called `pulsio`, but I cannot find it anywhere, including the CircuitPython 6 library bundle.

dhalbert, 2021-01-20:
`pulseio` is a built-in (C) module, but is not available on SAMD21 boards that don't have external flash chips ("Express" boards), for space reasons.

You could make a custom build, at the expense of removing some other things that you might not need. Or just switch to a more capable Express board.

dhalbert, 2021-01-20:
What's on which board: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html

obiben, 2021-01-22:
Thanks @dhalbert, that matrix clearly shows I'm either dyslexic or an idiot. `pulseio` is included on the M0, `pulsio` just doesn't exist!

closed 2021-01-22 by obiben