adafruit/circuitpython #4030: esp32s2: microcontroller.nvm "NVS Error"

jepler, 2021-01-21:
I noticed and @anecdata confirmed that if the first operation on nvm is a _read_, you'll get an "NVS error":
```
>>> import microcontroller
>>> microcontroller.nvm[0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: NVS Error
>>> len(microcontroller.nvm)
20480
>>> microcontroller.nvm[0] = 5
>>> microcontroller.nvm[0]
5
```

microDev1, 2021-01-21:
This is the expected behavior on _esp32s2_ port... the flash needs to be written before reading it.
The `NVM` implementation here doesn't directly interact with the flash instead it is built on top of [`NVS`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/storage/nvs_flash.html) library.

microDev1, 2021-01-21:
The `ByteArray` functionality is emulated. `NVS` uses key-value pairs... in our case :-
```python
microcontroller.nvm[a] = b
# a is the key
# b is the value
```
If a key isn't found... we throw an error `RuntimeError: NVS Error`. This however can be avoided by instead returning `0`.

jepler, 2021-01-21:
I encountered a second problem which may be related to the same design choices, so I'm adding it to this issue.

I wanted to assign each element in microcontroller.nvm, there are apparently 20480 of them.  However, after assigning 503 of them I encountered an exception:
```
Adafruit CircuitPython 6.1.0-rc.1-16-g89d716507-dirty on 2021-01-20; Kaluga 1 with ESP32S2
>>> len(microcontroller.nvm)
20480
>>> for i in range(len(microcontroller.nvm)): microcontroller.nvm[i] = i % 256
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to write to nvm.
>>> i
503
```
I am guessing that since each element I assign uses up a key and a value, that it quickly uses up the available space. The nvs "partition" is 20480 bytes in a 4MB device, but each key-value pair must be taking approximately 40 bytes to run out after 503 of them are added.

jepler, 2021-01-21:
At this point, nvm seems unwritable, even at indices that were written before:
```
>>> microcontroller.nvm[0] = 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to write to nvm.
```

microDev1, 2021-01-21:
I am able to replicate this... though for some reason the max key caps out at `403` for me.

```python
Adafruit CircuitPython 6.1.0-rc.1-16-g89d716507 on 2021-01-21; microS2 with ESP32S2
>>> len(microcontroller.nvm)
20480
>>> for i in range(len(microcontroller.nvm)): microcontroller.nvm[i] = i % 256
... 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unable to write to nvm.
>>> i
403
```
 
This was a concern when implementing `NVM` as such but it was considered a good trade off between wear-leveling functionality and utilization of available storage. It might be possible to optimize this though... with the current implementation the keys are stored as strings. A compression algorithm here could save a lot of space.

dhalbert, 2021-07-15:
We can fix this so that referencing an unwritten key simply returns 0. In the long run we might want to rewrite this using, say, a single NVS key.

microDev1, 2021-07-15:
After giving this some more thought... I think the nvm implementation should be refactored to just do a sequential byte-array write to maintain consistency with other ports (nvm is only meant to provide basic flash write functionality) and this way the full 20480 bytes will be available.

jepler, 2021-08-25:
This may technically be an incompatible change, as changing the representation in NVS will make the data unreadable by older revisions. I'm going to go ahead and work on this, but we need to consider this wrinkle before deciding if we can still go into 7.0.0 with this change.

(it may be possible to migrate the data forward but that would still make it inaccessible to 6.3)