adafruit/circuitpython #4034: Missing GP15 in board module for Raspberry Pi Pico

waiweng83, 2021-01-21:
CircuitPython version: 6.1.0-rc.1-51-gfb1e0106b

dir(board) yield the output of:
['__class__', 'A0', 'A1', 'A2', 'GP0', 'GP1', 'GP10', 'GP11', 'GP12', 'GP13', 'GP14', 'GP16', 'GP17', 'GP18', 'GP19', 'GP2', 'GP20', 'GP21', 'GP22', 'GP25', 'GP26', 'GP26_A0', 'GP27', 'GP27_A1', 'GP28', 'GP28_A2', 'GP3', 'GP4', 'GP5', 'GP6', 'GP7', 'GP8', 'GP9', 'LED']

There are missing GP15 in the list.

ladyada, 2021-01-21:
This is on purpose, GP15 should not be used, it is used by the internal USB peripheral

jepler, 2021-01-21:
This is due to our understanding of Erratum RP2040-E5 -- see [the datasheet ](https://datasheets.raspberrypi.org/rp2040/rp2040_datasheet.pdf) down at page 632 or so.

waiweng83, 2021-01-25:
I have gone through the software fix as suggested by the Errata here:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c

If I didn't get it wrongly, what it does is temporary turn the GP15 into an input and override its value as 1 (So that it will not be affected by the actual state of the pin) for 1ms during USB device enumeration.

This shouldn't have any bad impact to the external hardware connected to GP15 as the will be floating (or pulled up or pulled down?). Furthermore, this only happens when the USB is plugged into the host and usually the GPIO has not been initialized yet during this time. GPIO that initialized after this will still function normally.

Anyway, I'm thinking the support for GP15 is very essential due to the following:
1. Raspberry Pi official micropython does support it.
2. The official guide book from Raspberry Pi Foundation - [Get Started with MicroPython on Raspberry Pi Pico](https://www.raspberrypi.org/documentation/pico/getting-started/) has examples of using GP15 (eg: LED on Page 52)
3. The VGA example in the official "[Hardware Design with RP2040](https://datasheets.raspberrypi.org/rp2040/hardware_design_with_rp2040.pdf)" is using GP15 too for its VGA output.

I really hope that the support for GP15 will be added in the future release, or at least there is a workaround to enable it.

Thanks.

dhalbert, 2021-01-25:
We are checking whether the errata fix interferes with later use or not, as you suggested. In the meantime, you can use an alternate name for the pin, for example
```python
import digitalio
import microcontroller

gp15 = digitalio.DigitalInOut(microcontroller.pin.GPIO15). # this is the same as what board.GP15 would be
```

tannewt, 2021-01-25:
> I have gone through the software fix as suggested by the Errata here:
> https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c

My concern with the current fix is that it overrides the mux setting but doesn't set it back after. So, you could be using a pin and it will stop working. Since this is a USB reset, CircuitPython will likely auto-reload the VM anyway though.

> 2. The official guide book from Raspberry Pi Foundation - [Get Started with MicroPython on Raspberry Pi Pico](https://www.raspberrypi.org/documentation/pico/getting-started/) has examples of using GP15 (eg: LED on Page 52)

We're working on a CircuitPython Learn Guide for the Pico that mirrors the book's examples. (The examples will be added [here](https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython).) Our examples don't use GP15 because of this errata and we plan on keeping it that way even if we can use GP15.

> I really hope that the support for GP15 will be added in the future release, or at least there is a workaround to enable it.

We're working on it in #4063 and it'll likely be released as 6.2.0-beta.1 in the next day or two.

> We are checking whether the errata fix interferes with later use or not, as you suggested. In the meantime, you can use an alternate name for the pin, for example
>
> ```python
> import digitalio
> import [...]

This shouldn't work. The pin will be marked as in use.

waiweng83, 2021-01-26:
> My concern with the current fix is that it overrides the mux setting but doesn't set it back after. So, you could be using a pin and it will stop working. Since this is a USB reset, CircuitPython will likely auto-reload the VM anyway though.

If there is such concern, the pin settings can be saved prior to calling the fix function and restore it after. By the way, I'm not sure whether the fix is implemented because I can't seem to find where the function _rp2040_usb_device_enumeration_fix_ is called from, in both the micropython and circuit python.

> We're working on a CircuitPython Learn Guide for the Pico that mirrors the book's examples. (The examples will be added here.) Our examples don't use GP15 because of this errata and we plan on keeping it that way even if we can use GP15.

That's great. I personally do prefer circuitpython over micropython due to its huge libraries and supports from the community.

> We're working on it in #4063 and it'll likely be released as 6.2.0-beta.1 in the next day or two.

Thanks. I'll check it out and revert if there is any problem.

> This shouldn't work. The pin will be marked as in use.

I tested it and actually it's working. The only concern is if GP15 is not initialized in the code, it will be output with logic high by default.

kevinjwalters, 2021-01-26:
I'll mention @ghalfacree and @benevpi the authors of the [Get started withMicroPython on Raspberry Pi Pico](https://hackspace.raspberrypi.org/books/micropython-pico) book which has the pin 15 example in case they're interested in this.

benevpi, 2021-01-26:
We used pin 15 in the book before it was being used by the USB fix. The USB fix doesn't interfere with the examples we use, but yes, best to avoid if possible.

Ben

closed 2021-01-26 by dhalbert

tannewt, 2021-01-26:
> I tested it and actually it's working. The only concern is if GP15 is not initialized in the code, it will be output with logic high by default.

It should be reset like all other pins before the VM runs and after.

aallan, 2021-01-29:
The fix is only used for less than 1ms during enumeration and power on, for CircuitPython it’s unlikely you’ll have another serial port running — and another serial port is the main trigger for the bug. Also for basic GPIO it’s probably not an issue at all, it would only really be a problem if it’s assigned to a peripheral.

The fix will not output anything to the pin, but it will stop the output path from an active peripheral for a short period. The function select changes both the output and input path, which we need to change to USB. We then force the input path in software to a 1 regardless of what is on the pin. In the case of I2C clock or SPI TX that could potentially cause data to be lost.

Should be fine to enable in CircuitPython in other words.

tannewt, 2021-01-29:
@aallan I think you are assuming that USB is init on power up. That's not generally true though because we'll start to see battery powered boards where USB is connected later. (CircuitPython is likely to restart user code after USB starts though so the impact will be short lived.) I think the fix could be improved to reset the pin function back to the original value. As-is I don't think it changes it afterwards.

codemicro, 2021-02-01:
Should this be mentioned in docs somewhere (maybe [here](https://circuitpython.readthedocs.io/en/latest/ports/raspberrypi/README.html)), or is this a temporary thing? It's caused me some issues on `6.2.0-beta.1` and I feel like it'd be helpful for it to be documented if it's going to be a bug that sticks around for a while.

ladyada, 2021-02-01:
we added a faq/troubleshooting page here
https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython/faq-and-troubleshooting

kevinjwalters, 2021-02-04:
An interesting issue using `GP15`, [Adafruit Forums: Code.py Startup Script Not Starting on Raspberry Pi Pico](https://forums.adafruit.com/viewtopic.php?f=60&t=174992).

tannewt, 2021-02-04:
Here is a pending PR to restore the pad state: https://github.com/raspberrypi/pico-sdk/pull/80

PR #4063 (2021-01-26): RP2040: add GP15
Fixes #4034.

It looks like GP15 should be available after USB reset for general use.